realsenseai/librealsense #94: Narrowing conversion in v0.9.1 - Found using clang

vidstige, 2016-04-04:
Using v0.9.1 I cannot build using clang. I get the following output. It would be great if narrowing conversions where handled more explicitly. It looks like there is no bug in the code at hand, but in general I think it's good to avoid doing to crazy things to keep things working nicely.

/usr/include/libusb-1.0/libusb.h:732:3: warning: zero size arrays are an extension [-Wzero-length-array]
        [0] /* non-standard, but usually working code */
         ^
/usr/include/libusb-1.0/libusb.h:763:3: warning: zero size arrays are an extension [-Wzero-length-array]
        [0] /* non-standard, but usually working code */
         ^
/usr/include/libusb-1.0/libusb.h:1257:3: warning: zero size arrays are an extension [-Wzero-length-array]
        [0] /* non-standard, but usually working code */
         ^
src/uvc-v4l2.cpp:164:43: error: non-constant-expression cannot be narrowed from type 'int' to '__u8'
      (aka 'unsigned char') in initializer list [-Wc++11-narrowing]
                uvc_xu_control_query q = {xu.unit, control, UVC_GET_CUR, static_cast<uint16_t>(size), re...
                                          ^~~~~~~
src/uvc-v4l2.cpp:164:43: note: insert an explicit cast to silence this issue
                uvc_xu_control_query q = {xu.unit, control, UVC_GET_CUR, static_cast<uint16_t>(size), re...
                                          ^~~~~~~
                                          static_cast<__u8>( )
src/uvc-v4l2.cpp:170:43: error: non-constant-expression cannot be narrowed from type 'int' to '__u8'
      (aka 'unsigned char') in initializer list [-Wc++11-narrowing]
                uvc_xu_control_query q = {xu.unit, control, UVC_SET_CUR, static_cast<uint16_t>(size), re...
                                          ^~~~~~~
src/uvc-v4l2.cpp:170:43: note: insert an explicit cast to silence this issue
                uvc_xu_control_query q = {xu.unit, control, UVC_SET_CUR, static_cast<uint16_t>(size), re...
                                          ^~~~~~~
                                          static_cast<__u8>( )
3 warnings and 2 errors generated.
make: *** [obj/uvc-v4l2.o] Error 1

ddiakopoulos, 2016-04-04:
Can you be specific about what version of clang you are on? We see narrowing conversion errors from time to time, but our continuous integration infrastructure usually lets us know. Between Travis and Appveyor, every commit gets built on clang, GCC, and the vs2013 compiler.

vidstige, 2016-04-04:
Cool, yeah I saw you also build on Mac. Very nice indeed. Anyway this was
on clang 3.6

sgorsten, 2016-04-04:
Ah, you've exposed a gap in our testing. We currently only test the V4L2 backend on gcc, the LibUVC backend on clang, and the WMF backend on Visual Studio, which lets stuff like this sneak in sometimes.

sgorsten, 2016-04-04:
Should now be fixed in 2b19cde89b42ff325d407e4822d4b8ee8e149ee4

closed 2016-04-04 by sgorsten

vidstige, 2016-04-05:
Great! Thanks a lot for the quick fix guys. :-) 

Short git hint of the day: You can easily rebase your branch to make the history cleaner and get rid of the "merged in master into my branch"-commits.