realsenseai/librealsense #96: Camera temperature read-outs?

vidstige, 2016-04-05:
If I remember correctly, you could in the Windows SDK query the camera temperature. I only found stuff relating to temperature in the f200-private code. But can I get temperature readouts from a SR300 camera? If so, how would I go about that. I'm not afraid to look at the f200 code and try to recreate the feature for SR300....

dorodnic, 2016-04-05:
I believe SR300 reports temperatures using same API as F200. Check GetIRTemp = 0x52 and GetMEMSTemp = 0x0A USB commands and the related read_ir_temp and read_mems_temp functions. 
There are couple of changes that might be relevant: First, when camera is not streaming depth, F200 would successfully return temperature 0, while SR300 will fail with 'Invalid HW State" code. 
Another difference is that the temperature control loop in F200 is done in software, while SR300 is handling that automatically in hardware.

vidstige, 2016-04-05:
Alright, thanks I'll try that!

vidstige, 2016-04-19:
What is the difference between mems_temp and ir_temp? Which one should I use? Should I expose both?

vidstige, 2016-04-19:
I'm I'm getting ir temps ~29 and mems temps around ~20. What units are these? Why are they different?

ddiakopoulos, 2016-04-19:
@vidstige are you building your own temperature/calibration model for the SR300?

vidstige, 2016-04-19:
Nope, just want the read outs. We're using the pcb and think we got enough cooling, but are not really sure. Things seems to work okay, but we would like to log the temperature to be able to see if we got heating problems if a camera fails or such.

ddiakopoulos, 2016-04-19:
I'll let @dorodnic fill in the knowledge gaps here. When I see 29 and 20 I don't see how it could be any other range than degrees centigrade.

vidstige, 2016-04-20:
Yeah, I think you mean degrees Celsius. Could be some "raw" format you need to shift one bit left or multiply with 1.2 or whatever as well. Just wanted to double check. Thanks for pinging @dorodnic

vidstige, 2016-04-20:
A related question for you @ddiakopoulos : If I were to make a pull request for this, what would be a good way to add it to the API? I'm thinking a function in the `device` class, e.g. `get_temperature` that takes an enum for either mems or ir temp. Any thoughts on that?

vidstige, 2016-04-22:
ping. any news on this? If no answer I'll propose a pull request and assume degrees Celcius.

ddiakopoulos, 2016-04-22:
It should come through the get_option and set_option system.

vidstige, 2016-04-22:
But is it really an "option" like the others? You can't really set it for example, and it will very for each call, so it is a diffrent kind of creature compared to the options that are in there now... But, I'll do it like that. Thanks for taking your time to think about his and for the feedback!

dorodnic, 2016-04-24:
Hi vidstige, sorry for late response. 

Yes, the temperature is in degree Celsius.
Regarding the different functions - you can query the temperature on different points on the PCB:
ir_temp is measured near the IR Sensor.
mems_temp is measured near the projector.
These test points are device specific, and in theory new test points can be added for future products.

I fear that exposing temperature as an option abuses option API. 
Current API provides no way to know which options are read-only and which are read-write. Also, there is no way to know which options are safe to cache and which (like temperature) are being modified by the device. Since monitoring temperature is common use case, I would recommend adding get_temperature method.

ddiakopoulos, 2016-05-14:
Hi @vidstige We'll add this to an internal feature tracker because it's functionality that should be exposed. I'll back out of my comment that it should be built on the get/set option API and defer to @dorodnic for a possible implementation.

vidstige, 2016-05-16:
Alright, great. Thanks for the clarification and for looking into this. I got side tracked myself, was intending to make a pr for this but never got around.

digitologist, 2016-11-01:
Checking in on this request, have also come across this need.  Any updates?

vidstige, 2016-11-01:
No, it turned out we did not really need this as badly as we thought. We're not currently working on it and it is not planned. It did not seem too time consuming too implement last time I checked the code though.

closed 2017-08-10 by dorodnic