realsenseai/librealsense #97: cpp-config-ui crashes when changing some parameters.

jvgomez, 2016-04-06:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.04 | 
| Operating System & Version |  Ubuntu 14.04 64Bits |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic |
| Build System | Makefile | 

Hi, I have the R200 properly working (I can see all the streams and they make sense). However, `cpp-config-ui` crashes when I try to change some values. For instance:

color_white_balance:
```
rs.warn: Camera head contents version != 12, data may be missing/incorrect
RealSense error calling rs_set_device_option(device:0xe50670, option:COLOR_WHITE_BALANCE, value:6445):
     VIDIOC_S_CTRL error 5, Input/output error

```

or r200_depth_units:
```
RealSense error calling rs_set_device_option(device:0x27b5670, option:R200_DEPTH_UNITS, value:1):
     UVCIOC_CTRL_QUERY:UVC_SET_CUR error 5, Input/output error
```

Additionally, I always get the following warnings when starting the capture:
```
rs.warn: Camera head contents version != 12, data may be missing/incorrect
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
```

When I run `cpp-capture` I only get:

`rs.warn: Camera head contents version != 12, data may be missing/incorrect`

Is this normal or I have something wrong? Thanks for your help!

ddiakopoulos, 2016-04-06:
This is normal and expected. Some parameters can only be set before/after starting streaming. A couple controls documented in other issues (color auto exposure on Linux) are fundamentally broken.cpp-config-ui is a somewhat incomplete example in that we don't have restrict what controls can be set at what time. The bad magic numbers happen on occasion... usually points to USB xfer issues. If they go away after a few frames then no worries.

jvgomez, 2016-04-06:
Thanks for the quick reply! Everything clear now :)

closed 2016-04-06 by jvgomez