realsenseai/librealsense #100: Colour rectification extrinsics

ChrisN-987, 2016-04-07:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version 1.0.76 |   | 
| Operating System & Version |  Windows 10 |
| Build System |  VS2013 | 

Following on from a previous question I am now trying to rectify my captured RGB images offline. I have stored the camera intrinsics for the unrectified and rectified streams and I am now trying to emulate the activity in src/image.cpp to rectify a frame for a particular camera without having it connected.

The first step is computing the rectification table which has 3 parameters.
1: get_intrinsics()
2: get_extrinsics_to(source)
3: source.get_intrinsics()

For parameter 1 I am giving the unrectified colour stream intrinsics.
For parameter 3 I am giving the rectified colour stream intrinsics. 

I am confused about parameter 2. Is this just the identity matrix? Or do I need to also store the extrinsics of unrect - > rect? Also is my approach sensible? I am also storing extrinsics of depth->rect colour so I can align the depth offline too.

Thanks!

ChrisN-987, 2016-04-07:
I computed the extrinsics for colour->rectified colour as well. It seems there is some rotation. I feel like I am almost there. I am now trying to convert an OpenCV mat into (rsimpl::byte*) that the align image function requires without much success.

ChrisN-987, 2016-04-08:
Okay so I think I managed to get something using the compute rectification table followed by rectify image. I am getting the following results:

unrectified: http://s15.postimg.org/q8mdkh4x7/unrectified.jpg
rectified: http://postimg.org/image/osnv3fwup/

Notice the blank area in the top right of the image?

The only thing I am doing different to librealsense is I am not calling resize after computing the rectification table. Fristly this is because I am using a byte array rather than a byte vector and that the size of rectified and unrectified is the same so I am not sure I need to resize.

ChrisN-987, 2016-04-08:
I have managed to solve it. My issue was I was using the extrinsics of color->rectified color in the rectification table computation. It feels counter intuitive but the rectification table function actually needed rectified color->color.

closed 2016-04-08 by ChrisN-987

ddiakopoulos, 2016-04-08:
Sorry we didn't have time to chime in with any helpful hints... but in any case, glad it works!