realsenseai/librealsense #101: Is the depth interpolation not available on F200

bpwiselybabu, 2016-04-07:
| Required Info  |   |
|---|---|
| Camera Model | F200 | 
| Firmware Version | 2.38.5.0   | 
| Operating System & Version | Ubuntu 14.04   |
| Kernel Version (Linux Only)  |  3.19.0-49-generic |
| Build System |  Makefile | 

I noticed that the f200 depth map at 640x480, has holes. Is there any patch or update that has the depth map from f200 filled.

ddiakopoulos, 2016-04-07:
You can attach screenshots to Github issues. I recommend visually explaining the issue by attaching a PNG or JPEG.

bpwiselybabu, 2016-04-07:
[image]

ddiakopoulos, 2016-04-07:
Looks normal to me. The 'grid' is opengl rendering artifacts.

closed 2016-04-07 by ddiakopoulos

bpwiselybabu, 2016-04-07:
It is not GL rendering defect, I dont understand why you closed it so hastily.

I loaded the depth image, then I marked all the places with actual zero in the depth image with a white. If it was a visualization issue this should not happen because I am writing it to file.

[image]

richardw347, 2016-04-07:
I can confirm this I have the same issue with my F200: 

[image]

For comparison this is the registered depth image from an asus xtion (not running at the same time to avoid interference):

[image]

bpwiselybabu, 2016-04-07:
I suspect if they are going to look at it, but from what I understand the depth interpolation is not working correctly because in F200 we get both depth and rgb with the same resolution of 640x480 but their FOV are different. The alignment code has some issues with this.

ddiakopoulos, 2016-04-07:
@bpwiselybabu @richardw347 you are correct and @sgorsten will push a fix soon. Maybe in a few minutes! I didn't understand that you were talking about an interpolated image at first.

sgorsten, 2016-04-07:
This should be fixed in 90f1b2684e6e815191caccda70c4ca17a61ceced.

richardw347, 2016-04-07:
Looks good, thanks for the really quick response :)
[image]