realsenseai/librealsense #98: Windows 10 Pro Insider Preview fails in uvc-wmf.cpp

bobdavies2000, 2016-04-07:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | Windows 10 pro Insider Preview  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013| 

Every application gets the following error:

RealSense error calling rs_create_context(api_version:4):
    IKsControl::KsProperty(...) returned 0x80004003

The camera was working fine on Windows 10 before I upgraded.  I may have made a mistake upgrading to the Windows 10 Pro Insider Preview.  All 3 subdevices are present in the device manager.  I tried cpp-capture but all the librealsense apps fail with this error.  The 0x80004003 is a problem with E_POINTER - bad reference provided.  It fails at "rs::context ctx;"  The failure occurs with the call:

auto it = ks_controls.find(xu.node);

with it = 0xcdcdcdcd upon return with xu.node = 1

I can back off the preview edition...

ddiakopoulos, 2016-04-07:
@bobdavies2000 makes me question if this is a librealsense issue or a Windows issue. It works alrighty on vanilla Windows 10. It's possible they changed implementation details of WMF in the Insider Preview, but also possible that it is just broken in that version.

bobdavies2000, 2016-04-07:
Thanks for the note.  I just went back to Windows 10 and with no changes to
the hardware, everything is working fine.  They have broken the uvc-wmf
interface in that preview.

Bob

closed 2016-04-08 by ddiakopoulos

sergioamr, 2016-05-30:
I can confirm that this problem exist on W10 Build 14352.