realsenseai/librealsense #103: crash when running xcode examples

changizi, 2016-04-08:
I have tried running librealsense xcode examples on OS X system with R200 camera. They usually work but sometimes they crash with this error:

malloc: *** error for object 0x100619078: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug

Has anyone experienced this?

narner, 2016-04-08:
I'm experiencing the same behavior. I'm also using an R200 camera on OS X (10.11.3).

qianyizh, 2016-04-08:
Same here.

Intel RealSense R200
Serial number: 2411006589
Firmware version: 1.0.72.06

OS X 10.11.4
LLVM 7.3 (or XCode 7.3, it's the same)

Crashes even on very simple code, e.g.:
int main(int argc, char **args)
{
    rs::context ctx;
    return 1;
}

Sometimes it works fine. Sometimes it reports the error when context is being deleted.
Seems to be an issue with libuvc/libusb.

ddiakopoulos, 2016-04-09:
Is the issue only on shutdown of the context or does the app crash in the middle of streaming?

changizi, 2016-04-09:
For me it usually happens when the start function is called on the device. But again it's not consistent. Sometimes it works perfectly.

narner, 2016-04-09:
For me, the app would crash in the middle of streaming.

qianyizh, 2016-04-09:
I haven't done extensive tests on streaming yet.
The issue happens on shutdown of the context.
Sometimes it happens in the call of:
  if (ctx->own_usb_ctx)
    libusb_exit(ctx->usb_ctx);
Yet it is not consistent.

ddiakopoulos, 2016-04-09:
Alright! I'll spend some time this weekend testing out on a few different OSX configurations. Needless to say, sounds like a software bug in librealsense so I'll try to track it down.

qianyizh, 2016-04-09:
I managed to reproduce the error on another Macbook. Here is the workflow.
(OS X 10.11.4, XCode 7.3)

1. Install Homebrew

2. Use Homebrew to install libusb, pkg-config, homebrew/versions/glfw3

3. Use XCode to open librealsense.xcworkspace

4. Set launch app to cpp-tutorial-1-depth

5. Change code of cpp-tutorial-1-depth.cpp to
`
#include <librealsense/rs.hpp>
#include <cstdio>

int main(){
    rs::context ctx;
    return 1;
}
`

6. Run cpp-tutorial-1-depth for a few times (keep pressing command+R)

7. You will see the error after a few runs.

> cpp-tutorial-1-depth(6343,0x1001b9000) malloc: *** error for object 0x1003002f8: incorrect checksum for freed object - object was probably modified after being freed.
> *** set a breakpoint in malloc_error_break to debug
> Program ended with exit code: 9

It doesn't matter if the R200 sensor is plugged in or not.

ddiakopoulos, 2016-04-09:
@changizi @narner @qianyizh I just pushed a commit bfeed0e65da3de8d1a2859c27986205010ec219c. Can you test with the latest?

narner, 2016-04-09:
So far things seem to have improved (the crash would usually happen a few
seconds after starting one of the programs).

qianyizh, 2016-04-09:
I haven't seen the issue after I applied the patch. Tested on both my Macbooks.

ddiakopoulos, 2016-04-11:
@changizi waiting on your confirmation that this fixes the issue so I can close it out :)

changizi, 2016-04-11:
I get this error after testing the latest:
Using device 0, an Intel RealSense R200
    Serial number: 2431003254
    Firmware version: 1.0.72.06
rs::error was thrown when calling rs_start_device(device:0x100804c18):
    uvc_set_ctrl(...) returned LIBUSB_ERROR_PIPE

ddiakopoulos, 2016-04-11:
@changizi can you try a few more times? That error seems unrelated to the memory issue at hand

changizi, 2016-04-11:
Looks like the memory issues is gone. Thanks for the fix.
Do you know why I am getting the above error though? If I disconnect and reconnect the sensor before I run each example it runs fine, otherwise I get that error.

ddiakopoulos, 2016-04-11:
@changizi yeah the unplug/replug on OSX with the libuvc backend is probably the oldest bug we have yet to track down. It is documented in #31 and remains open...

blackerpaper, 2016-04-13:
@changizi
This issue is linked to the "replug" issued. It's closed now, seems the bug comes from the osx usb driver.

Solution: Just update osx to the newest one (10.11.4) and this issue gone.

Hope this will help.

ddiakopoulos, 2016-04-19:
Closing this out since the memory error has been fixed and disconnection issues are documented in a different thread!

closed 2016-04-19 by ddiakopoulos