realsenseai/librealsense #105: Compatible firmware for F200 to work with librealsense

kcINual, 2016-04-11:
From the intro of librealsense, the F/W version for F200 to work with this API is 2.60.0.0

But the latest F/W that I can find on Intel (and then directed to [Creative](http://support.creative.com/Products/ProductDetails.aspx?catID=218&CatName=Web+Cameras&subCatID=231&subCatName=MIDI+Keyboards&prodID=22257&prodName=Intel+RealSense+Developer+Kit&bTopTwenty=1&VARSET=prodfaq:PRODFAQ_22257,VARSET=CategoryID:218&)) is 2.38.5.0 released on 2014 Nov 19.

Wondering where can I download the F/W 2.60.0.0 for F200?

Successfully compiled librealsense and its examples, but got the error of **"No device detected. Is it plugged in?"** Wondering if it's the F/W version that causes this error. I'm running on Intel's NUC NUI5i3RYH with Intel HD Graphics 5500.

Thanks!

| Camera Model | F200  | 
| Firmware Version | 2.38.5.0  | 
| Operating System & Version | Ubuntu 14.04.01  |
| Kernel Version (Linux Only)  | 3.19.0  |
| Build System |  Makefile |

ddiakopoulos, 2016-04-11:
Hi @kcINual please read the section on Firmware Update in the Readme: https://github.com/IntelRealSense/librealsense/blob/master/readme.md You will need a Windows 10 machine and install some other Intel software in order to update the firmware to the latest available.

On the other hand, I'm not sure updating firmware would help with your `No device detected` issue. What is the output of `lsusb` on your machine? 

Also curious about your choice of the 3.19.0 kernel. Did you read our installation guide? We tried to make it clear that only 4.2 and above is normally supported.

kcINual, 2016-04-12:
Thanks @ddiakopoulos ! 

Just have a clean install of Ubuntu 14.4 with kernel `4.2.0-27-generic` and then `4.2.0-35-generic`. But the problem persists.

After installing video4linux (without plugging in the F200):

```
$ sudo dmesg | tail
[  166.051452] media: Linux media interface: v0.10
[  166.059042] Linux video capture interface: v2.00
[  166.068061] usbcore: registered new interface driver uvcvideo
[  166.068065] USB Video Class driver (1.1.1)
```

After plugging in the F200:

```
$ sudo dmesg | tail
[  276.079844] usb 1-1.4: new high-speed USB device number 5 using ehci-pci
[  276.174497] usb 1-1.4: New USB device found, idVendor=05e3, idProduct=0610
[  276.174502] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  276.174504] usb 1-1.4: Product: USB2.0 Hub
[  276.174505] usb 1-1.4: Manufacturer: GenesysLogic
[  276.175031] hub 1-1.4:1.0: USB hub found
[  276.175467] hub 1-1.4:1.0: 2 ports detected
[  276.839860] usb 1-1.4.2: new full-speed USB device number 6 using ehci-pci
[  277.090094] usb 1-1.4.2: New USB device found, idVendor=041e, idProduct=4099
[  277.090098] usb 1-1.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  277.090101] usb 1-1.4.2: Product: Creative VF0800
[  277.090102] usb 1-1.4.2: Manufacturer: Creative Technology Ltd
[  277.090104] usb 1-1.4.2: SerialNumber: 2015XXXXXXXX
[  277.175683] input: Creative Technology Ltd Creative VF0800 as /devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2:1.3/0003:041E:4099.0003/input/input16
[  277.228621] hid-generic 0003:041E:4099.0003: input,hiddev0,hidraw2: USB HID v1.11 Device [Creative Technology Ltd Creative VF0800] on usb-0000:00:1d.0-1.4.2/input3
[  277.306798] usbcore: registered new interface driver snd-usb-audio
```

```
$ sudo ls /dev/vid*
ls: cannot access /dev/vid*: No such file or directory
```

Not sure if it's the cause of the `No device detected` issue -- I encountered an error when running `./scripts/patch-uvcvideo-ubuntu-mainline.sh`
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Picking 'linux-lts-wily' as source package instead of 'linux-image-4.2.0-27-generic'
The following packages have unmet dependencies:
 libudev-dev : Depends: libudev1 (= 204-5ubuntu20.18) but 204-5ubuntu20.19 is to be installed
```

This unmet dependency issue is solved by:
1. Adding these two lines in `/etc/apt/sources.list`
```
deb http://us.archive.ubuntu.com/ubuntu/ trusty-updates main    
deb-src http://us.archive.ubuntu.com/ubuntu/ trusty-updates main
```
2. Run `sudo apt-get update` and then `sudo apt-get install libudev-dev`. Rerun `./scripts/patch-uvcvideo-ubuntu-mainline.sh`

But the F200 is still not detected when running the example in `/bin/`. Tested with two F200 and the problem is the same. Both F200s' F/W are [confirmed](https://software.intel.com/en-us/forums/realsense/topic/562071) to have F/W 2.60.0.0 by running Intel's DCM (`intel_rs_dcm_f200_1.4.27.41944.exe`) on Win10.

teknotus, 2016-04-12:
The dmesg indicates it is finding the stereo microphones on the F200 but
not the cameras. Usually this means it isn't on a USB 3.0 port, or the port
doesn't provide enough power for the camera. Often only some ports on
computers have enough power, so trying a different port might work.

kcINual, 2016-04-13:
Thanks @teknotus ! Added extra power to the USB 3.0 port and now the F200 can be detected.

```
$ sudo ls /dev/vid*
/dev/video0  /dev/video1
```

Successfully ran a couple of examples in `/bin` but the following errors appear quite often.

```
$ ./cpp-restart 
rs.error: TemperatureControlLoop: opcodes do not match
rs.error: TemperatureControlLoop: opcodes do not match
```

```
$ ./c-tutorial-2-streams 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense F200
    Serial number: 02015XXXXXXX
    Firmware version: 2.60.0.0
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)
```

```
$ ./cpp-tutorial-2-streams 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense F200
    Serial number: 02015XXXXXXX
    Firmware version: 2.60.0.0
Device does not provide infrared2 stream.
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)
```

Sometimes the stream (including the RGB, depth, IR) appeared for a couple of seconds, and then the `VIDIOC_DQBUF error 19` show up.

Using `./cpp-config-ui`, the RGB stream works fine. Once the `Enable Depth` or `Enable IR` is checked and clicked `Start`, the `VIDIOC_DQBUF error 19` show up.

ddiakopoulos, 2016-04-13:
Hi @kcINual we recently did a round of debugging on an internal dev board and saw problems like you describe. As soon as the CPU/GPU was stressed, the device would forcefully disconnect with the `VIDIOC_DQBUF error 19` message. In our own case, it was fixed with a heap of modified settings in an unlocked BIOS related to power states and thermal limits. On a NUC similar to yours, I had similar USB 3.0 errors on different (non Intel) cameras that pointed to issues with the USB host controller.

In summary: seems like you successfully installed the lib on your system, but now you are brushing up against hardware issues that are (in my estimation) not worth tracking down and debugging. If possible, try a different machine?

kcINual, 2016-04-14:
Thanks @ddiakopoulos ! 

Now the F200 seems to work after I disabled the USB legacy support in BIOS. Now sure if it's the cure, and will report back the stability.

For those who use F200 along with Intel NUC and ubuntu, it may be a fix for the `VIDIOC_DQBUF error 19`.

ddiakopoulos, 2016-04-14:
@kcINual glad it's working! This is very helpful information.

closed 2016-04-14 by ddiakopoulos

ckim337, 2016-05-05:
@kcINual  When you added extra power to the USB 3.0 port, did you use the powerhub? 
If yes, can you please provide me the hub information? I am using SR300, and computer is not able to detect the camera.

kcINual, 2016-05-22:
@ckim337  What does the `dmesg | tail` say?

I didn't use a hub. Instead, I added +5V from a PSU (in which its ground is common to my NUC) to a USB extension cable, then connected my F200 to this cable.