realsenseai/librealsense #108: SR300 CAmera not working

aburbanol, 2016-04-13:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | dont known  | 
| Operating System & Version |  UBUTNTU 14.04 and Windows 8.1|
| Kernel Version (Linux Only)  | 4.4  |
| Build System |  Makefile | 

I have the F200 camera and is working normally, but when I connect my SR300 is not working. I tested in Linux and Windows, I install the new driver in windows. Any Idea?

ddiakopoulos, 2016-04-13:
@aburbanol so it is not working in either Linux or Windows where F200 was? sounds like a defective camera in that case

aburbanol, 2016-04-13:
@ddiakopoulos It shows like one frame and then:

terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device

aburbanol, 2016-04-13:
@ddiakopoulos So now I call the call center or something like that? btw Can You confirm the SR300 technology patent?  can we do it here? or has to be for another channel?

ddiakopoulos, 2016-04-13:
@aburbanol does it work on Windows 10 with the RealSense SDK? I can't comment on the other topic.

aburbanol, 2016-04-13:
My SR300 I havent tried in a windows 10, only in Ubuntu 14.4 k 4.4 with libsReal sense and in Windows 8.1 with RealSense SDK

ddiakopoulos, 2016-04-13:
@aburbanol OK. So does it work on Windows 8.1 with the RealSense SDK? I wasn't aware the SDK provided a DCM for the SR300 on Windows 8.1

aburbanol, 2016-04-14:
dear @ddiakopoulos as I said in the first message it is not working in any of the both OS that I put in the table ( ubuntu 14 and windows 10). I mention the other camera F200 just to tell you that I'm sure that the framework (librealsense or RealSense SDK) is working.

However, So no change to make it work? you think it has some problems?

aburbanol, 2016-04-18:
@ddiakopoulos Any test I can do to see if the problem is the camera or the librealsense? something I can do to help you to understand the problem?

ddiakopoulos, 2016-04-18:
@aburbanol I'm stumped! If F200 works and SR300 does not, it sounds like hardware issues and not something related to software. Assuming you ordered the SR300 from click.intel.com, you can request an RMA or replacement by emailing click.support@intel.com

aburbanol, 2016-04-19:
@ddiakopoulos Ok done! Let you know If a got a new one and if it works!

aburbanol, 2016-04-19:
@ddiakopoulos Dang!! I got confuse with the models and checking the camera boxes in order to send the support email, the camera that I have is a R200. IS working I hope It wont change anything! SORRY!

ddiakopoulos, 2016-04-19:
@aburbanol I'm seriously confused. You even wrote the correct firmware version in the issue table that only applies to SR300. Did you copy and paste that some the readme?

aburbanol, 2016-04-19:
@ddiakopoulos SR300 is the one that is not working, I make the mistake was about the camera's reference that works. I have the R200, no the F200, but in any case is the SR300 that is not working!!!

the camera that i wrote wrong is the one that is working and which I checked that the LibRealSense is working in my computer!

ddiakopoulos, 2016-04-25:
@aburbanol keep us updated when you receive a new SR300.

aburbanol, 2016-04-28:
@ddiakopoulos  sure Ill let you know! for the moment they dont have stuck so until mid-may they will send a new one! I think ill have it at the end of the month!

ckim337, 2016-05-05:
@ddiakopoulos I also have trouble connecting SR300. I am guessing that computer is not able to detect the camera. It worked at the beginning, then starts to disconnect after couple seconds. Do you if there is requirement power or ampere for SR300 to be detected in the computer? Error occurs saying that it cannot detect the Video4Linux. I have two SR300 camera, and neither one is working.
@aburbanol Have you got the new one ? did it work?

aburbanol, 2016-05-06:
@ckim337 I havent, They told me they get new cameras until mid may, then they will send me one... asap I have it, I will let you know!

ddiakopoulos, 2016-05-20:
@aburbanol have you gotten a replacement camera yet?

aburbanol, 2016-05-20:
@ddiakopoulos NOOOOOOOOOOOOOOOOOOOOOOOO!!! I'm wainting :'(!

aburbanol, 2016-06-09:
@ddiakopoulos Finally I receive my new SR300, but is not fully working! I can run the Tutorial-1-depth but It wont work with tutorial 2 and 3. Using the Configuration UI Example I can start the Color stream.

I was able also to start only the depth stream for few seconds ONLY when I change the parameters f200_filter_option and sr300_auto_option_range_enable_motion_versus_ram.

So what is next @ddiakopoulos ???

These are the errors
**Tuto 2**
> There are 1 connected RealSense devices.
> 
> Using device 0, an Intel RealSense SR300
>     Serial number: 609203001339
>     Firmware version: 3.10.10.0
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  VIDIOC_DQBUF error 19, No such device

**Tuto-3**

> There are 1 connected RealSense devices.
> 
> Using device 0, an Intel RealSense SR300
>     Serial number: 609203001339
>     Firmware version: 3.10.10.0
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  VIDIOC_DQBUF error 19, No such device
> Press <RETURN> to close this window...

**Other errors:**

> RealSense error calling rs_set_device_option(device:0x1184f80, option:SR300_AUTO_RANGE_MAX_LASER, value:0):
>     opcodes do not match
> Press <RETURN> to close this window...

> RealSense error calling rs_set_device_option(device:0x134a0b0, option:SR300_AUTO_RANGE_START_LASER, value:-298):
>     opcodes do not match
> Press <RETURN> to close this windo

> RealSense error calling rs_set_device_option(device:0x27ef280, option:SR300_AUTO_RANGE_MIN_MOTION_VERSUS_RANGE, value:-1788):
>     opcodes do not match
> Press <RETURN> to close this window...

ddiakopoulos, 2016-06-16:
Hi @aburbanol do you have any other computers to test on? Since we've ruled out camera hardware issues, the only other thing I can think is that your computer has a flaky/unsupported USB 3.0 host controller. I also recommend trying it though a _powered_ USB 3.0 hub.

Sammy222, 2016-07-11:
Hi  @aburbanol! I have same problems like you! I have a PC with an Intel i7-4500U, 8Gb Ram, Usb 3.0 and Win 10 x64. My camera is jus t detected sometimes from the system. When I try to start the SDK-Browser the camera is just detected sometimes and this case only for a few seconds. I try to ireinstall DCM and SDK a few times but without success. I write to click.support@intel.com and they tell me there are no solutions. Did you get a new camera for free? I

Sammy222, 2016-07-11:
I never tried out the camera (SR300) on Ubuntu 14.4. I have just Ubuntu 16.4 installed and I think it is not supported in this version, isn't it? 
I'm glad to here some ideas or inputs from you guys. Thanks

aburbanol, 2016-07-12:
Dear @Sammy222 I had to send my old camera to them and I got a new one. The first one was definitely not working, this new one present a different behavior, however, none of them worked in UBUNTU.

@ddiakopoulos I tried in several Ubuntu and different computers and in none of them is working, while the R200 is! I'm gonna try it in windows and let you know.

closed 2017-03-21 by ddiakopoulos

aburbanol, 2017-03-21:
@ddiakopoulos I tried again recently, I pull the latest version of librealsense, I used the config-gui in order to get the depth stream or the color stream, but it is not working when I ask for both of them. The other examples are not working. Only the first one that again only use one stream.