realsenseai/librealsense #111: Realsense R200 camera enabling on chromium-os v3.18 kernel

gs0622, 2016-04-14:
Just FYI that I successfully enabled R200 on chromium-os v3.18 kernel, and filed ticket for kernel patch-set, it is under reviewing, also leave here a note in case someone is interesting to use it on chromebook, no action is required to librealsense team.

chromium-os ticket:
https://bugs.chromium.org/p/chromium/issues/detail?id=600944

ddiakopoulos, 2016-04-14:
This is great! Thanks for the update.

ds-hwang, 2016-05-31:
That's great news!
How about v3.14 and v4.4?

Skylake uses v3.18, Broadwell uses v3.14 and Broxton uses v4.4. If only v3.18 is supported, only Skylake can support realsense.

gs0622, 2016-06-01:
The chromeos-v4.4 kernel is partially functional for R200 because of missing below commit:
https://chromium-review.googlesource.com/#/c/335130/

And this was tracking on
https://bugs.chromium.org/p/chromium/issues/detail?id=598263

It seems to be merged however been rejected later on because of the v.4.4 was in sort of integration phase.
I am not sure whether it is unlocked now, I will verify it first, perhaps re-upload again to respect Valentin's work.

As to v3.14, I don't have answer yet, I need to check on my BDW chromebook, yet it might need some integration work, I suppose.

gs0622, 2016-06-07:
No surprisingly, v3.14 on BDW doest not support R200 yet; either malfunction or crash; filed a cloned ticket of crosbug 618042 (https://bugs.chromium.org/p/chromium/issues/detail?id=618042), I will tinker it to enable R200 on my BDW chromebook.

closed 2017-03-21 by ddiakopoulos