realsenseai/librealsense #112: Suggest using Xenial kernel package for ubuntu 14.04

mdhorn, 2016-04-14:
Currently, the install directions for 14.04 suggest download a specific release of the Ubuntu 15.10 (Wily Werewolf) kernel. The concern here is that locks the user into a specific kernel release which is not being updated and may miss critical security patches.

Instead, suggest using 'apt' and using the Ubuntu 16.04 (Xenial Xerus) kernel:
`sudo apt-get install linux-generic-lts-xenial`
This kernel is currently 4.4.0-18-generic, and should be updated for security fixes.

blackerpaper, 2016-04-16:
Why not using libuvc backend? It doesn't need kernel space and seems portable.

It's just slightly slower then the kernel backend.

teknotus, 2016-04-16:
15.10 is going to be maintained for a few more months, and 16.04 isn't
being released for a few days. It's the sensible stable kernel to recommend
as of today. The way out of mainline kernel patches work you may have to
rebuild uvcvideo.ko every few weeks either way as Ubuntu puts out new
kernel packages frequently.

ddiakopoulos, 2016-05-14:
This will be fixed in a forthcoming release, 0.9.2.

closed 2016-05-14 by ddiakopoulos