realsenseai/librealsense #116: Depth streams from multiple cameras mixing

ymoslehSE, 2016-04-20:
| Required Info  |   |
|---|---|
| Camera Model | R200 
| Firmware Version |  1.0.71.06 
| Operating System & Version |  Ubuntu 14.04 
| Kernel Version (Linux Only)  | 4.4.0-040400-generic
| Build System |  Makefile 

Hello,

When connecting two R200 realsense devices to the same machine and streaming depth from both cameras we're observing the depth stream from camera 1 mixes with the depth stream of camera 2.

The issue is not observed when streaming RGB color.

Any ideas on what could be the potential cause? 
Please let me know if you need any more information.

Thank you.

mkhansen-intel, 2016-04-20:
Are you streaming them to the same topic? Make sure that in the launch file you are using for each one you are giving a unique name (ie. camera1, camera2, left_camera, right_camera) in addition to the serial number. Then you'll see each one publish a separate topic (/camera1/depth/image_raw), etc.

mkhansen-intel, 2016-04-21:
Whoops scratch my last reply. I didn't realize you're asking this on the librealsense github. I thought this was asked on the ROS Realsense github (I get emails whenever an issue gets filed for either one).

ymoslehSE, 2016-04-21:
Thanks for the response,

Perhaps a code snippet can help clarify how I am performing the streaming:

```
rs_context * rsCtx;
rs_device * dev[MAX_CAMS]; // 2 cams
int device_count;
...
device_count = rs_get_device_count(rsCtx, NULL);

// Enumerate 
for(int i = 0 ; i < device_count ; ++i) {

dev[i] = rs_get_device(rsCtx, i, &error);
rs_enable_stream_preset(dev[i], RS_STREAM_DEPTH, RS_PRESET_BEST_QUALITY, &error);
...
}
...
// Get frame data
const void * pixel;
for(int i = 0; i < device_count; ++i) {

      rs_wait_for_frames(dev[i], &error);
      
           // 2 streams, depth and color
       for(int j = 0; j < MAX_STREAMS; ++j) {
          
          pixels = 0;
          if( 0 == j)
               StreamNum = DepthStream; // 0
           else
              StreamNum = ColorStream; // 1

            pixel = rs_get_frame_data(dev[i], (enum rs_stream) (StreamNum), &intrinsic, 0 );
            
          /* Copy frame data
              Push frame data to vector and send data for streaming.
           */
       }
}
```
Once we stream the frame data we notice the depth streams for the cameras are mixing at intermittent intervals.

ymoslehSE, 2016-04-21:
Np thanks

ymoslehSE, 2016-04-21:
Np, thanks.

ddiakopoulos, 2016-04-21:
Interesting! I've never seen this behavior before. Can you attach the full program so we can attempt to replicate it? Before digging into the software, let's try a couple things: 

Step 1. Do you have a Windows machine handy that you can use to update the firmware to the minimum recommended version using the RealSense DCM?

Step 2. Replicating a librealsense build on another machine with different hardware to see if there's a delta in behavior.

ymoslehSE, 2016-04-21:
Hello,

Thanks for the response. So far we have tried several machines. For example, we've tried it on a machine running an i7 Sandy Bridge and an Intel NUC running an i5 Broadwell. On both of these machines we observe the depth mixing behavior.

I currently have access to a windows 8.1 Ivy Bridge-E machine. However, the recommended specs for the DCM tool shows that it requires a Skylake processor and Windows 10. Will there be any issues if I ran the DCM tool on my Win 8.1 machine?

Thanks again for your help!

ddiakopoulos, 2016-04-21:
@ymoslehSE what about the behavior in the exanoke cpp-multicam to rule out any possible bugs in your own app?

ymoslehSE, 2016-04-21:
Hello,

We've tried running the cpp-multicam example and we don't notice any issues with the depth streams mixing.
It looks it could be a problem on our end, but not sure why the mixing is only being observed with the depth streams and not the color streams. 

Our capture app is quite large but relatively straightforward. It captures from the realsense camera and sends the data via RTP over the network to the cloud for image processing. 

If helpful I can provide you the src?

 Thanks again for your help.

ymoslehSE, 2016-04-22:
After further analyses it appears that the bug is in our program. Thanks again for your help!

ddiakopoulos, 2016-04-22:
Glad to be of assistance :)

closed 2016-04-22 by ddiakopoulos