realsenseai/librealsense #117: No image with R200 and realsense_camera launch files

CruJon, 2016-04-20:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04.4  |
| Kernel Version (Linux Only)  | 4.4.0-040400  |
| Build System | Makefile | 

Hi there, 

I'm new with Realsense and ROS and just trying to get everything up and running.
When starting with the following command, there are a few errors that I have no idea on how to solve them:

`crujon:~/catkin_ws$ roslaunch realsense_camera realsense_r200_camera.launch 
... logging to /home/jon/.ros/log/19cef56e-070b-11e6-89ce-b0c090467a11/roslaunch-jon-6816.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jon:42623/

SUMMARY
========

PARAMETERS
 * /realsense_node/debug_depth_unit: False
 * /realsense_node/depth_cx: 320.0
 * /realsense_node/depth_cy: 240.0
 * /realsense_node/depth_frame_id: camera_depth_opti...
 * /realsense_node/depth_fx: 463.888885
 * /realsense_node/depth_fy: 463.888885
 * /realsense_node/depth_scale: 0.001
 * /realsense_node/depth_unit: 33.1
 * /realsense_node/depth_uv_enable_max: 800
 * /realsense_node/depth_uv_enable_min: 0
 * /realsense_node/ir_camera_info_url: 
 * /realsense_node/realsense_camera_type: Intel RealSense 3...
 * /realsense_node/rgb_camera_info_url: 
 * /realsense_node/rgb_frame_id: camera_rgb_optica...
 * /realsense_node/topic_depth_points_id: camera/depth/points
 * /realsense_node/topic_depth_registered_points_id: camera/depth_regi...
 * /realsense_node/topic_image_depth_raw_id: camera/depth/imag...
 * /realsense_node/topic_image_infrared_raw_id: camera/ir/image_raw
 * /realsense_node/topic_image_rgb_raw_id: camera/rgb/image_raw
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    camera_base_link (tf/static_transform_publisher)
    camera_base_link1 (tf/static_transform_publisher)
    camera_base_link2 (tf/static_transform_publisher)
    camera_base_link3 (tf/static_transform_publisher)
    realsense_node (realsense_camera/realsense_camera_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[realsense_node-1]: started with pid [6834]
process[camera_base_link-2]: started with pid [6835]
process[camera_base_link1-3]: started with pid [6836]
process[camera_base_link2-4]: started with pid [6849]
process[camera_base_link3-5]: started with pid [6865]

===================
realsense_camera_type = Intel RealSense 3D Camera R200
rgb_frame_id = camera_rgb_optical_frame
depth_frame_id = camera_depth_optical_frame
depth_unit = 33.099998
depth_scale = 0.001000
depth_fxinv = 0.002156
depth_fyinv = 0.002156
depth_cx = 320.000000
depth_cy = 240.000000
depth_uv_enable_min = 0
depth_uv_enable_max = 800
topic_depth_points_id = camera/depth/points
topic_depth_registered_points_id = camera/depth_registered/points
topic_image_rgb_raw_id = camera/rgb/image_raw
topic_image_depth_raw_id = camera/depth/image_raw
topic_image_infrared_raw_id = camera/ir/image_raw
debug_depth_unit = 0
rgb_camera_info_url = 
ir_camera_info_url = 
=======================

depthWithIRStream - NOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
if you want IR stream, please visit
http://solsticlipse.com/2015/03/31/intel-real-sense-3d-on-linux-macos.html
https://github.com/teknotus/depthview/tree/kernelpatchfmt

===========================================
Intel(R) RealSense(TM) 3D Camera lists

PCI: usb-0000:00:14.0-1
Serial: SN_2401002183
	/dev/video1
	/dev/video2
	/dev/video3
===========================================

use camera SN_2401002183
Directory /home/jon/catkin_ws/src/realsense_camera/data/uvmap/SN_2401002183 does not exist!!!!
video rgb name is /dev/video1
video depth name is /dev/video2

============== VIDIOC_QUERYCAP
/dev/video1 
   driver = uvcvideo
   card = Intel RealSense 3D Camera R200
   bus_info = usb-0000:00:14.0-1
   version = 0x00040400
   capabilities = 0x85200001
   device_caps = 0x05200001
   reserved[3] = [0x00000000, 0x00000000, 0x00000000]

======================== get
/dev/video1 
v4l2_pix_format
   width = 320
   height = 962
   pixelformat_value = 0x56595559
   pixelformat = YUYV
   field = 1
   bytesperline = 640
   sizeimage = 615680
   colorspace = 0
   priv = -17970434

======================== set
/dev/video1 
v4l2_pix_format
   width = 320
   height = 962
   pixelformat_value = 0x56595559
   pixelformat = YUYV
   field = 1
   bytesperline = 640
   sizeimage = 615680
   colorspace = 0
   priv = -17970434

============ VIDIOC_S_PARM
/dev/video1 
v4l2_streamparm
   timeperframe.numerator = 1
   timeperframe.denominator = 60

video rgb w,h - 320, 962

============== VIDIOC_QUERYCAP
/dev/video2 
   driver = uvcvideo
   card = Intel RealSense 3D Camera R200
   bus_info = usb-0000:00:14.0-1
   version = 0x00040400
   capabilities = 0x85200001
   device_caps = 0x05200001
   reserved[3] = [0x00000000, 0x00000000, 0x00000000]

======================== get
/dev/video2 
v4l2_pix_format
   width = 314
   height = 938
   pixelformat_value = 0x56595559
   pixelformat = YUYV
   field = 1
   bytesperline = 628
   sizeimage = 589064
   colorspace = 0
   priv = -17970434

======================== set
/dev/video2 
v4l2_pix_format
   width = 628
   height = 469
   pixelformat_value = 0x00000000
   pixelformat = 
   field = 1
   bytesperline = 1256
   sizeimage = 589064
   colorspace = 0
   priv = -17970434

============ VIDIOC_S_PARM
/dev/video2 
v4l2_streamparm
   timeperframe.numerator = 1
   timeperframe.denominator = 60

libv4l2: error turning on stream: Input/output error

VIDIOC_STREAMON error 5, Input/output error
open /dev/video2 error!!!!!!!!
[realsense_node-1] process has finished cleanly
log file: /home/jon/.ros/log/19cef56e-070b-11e6-89ce-b0c090467a11/realsense_node-1*.log
`

Sometimes, with the same command, I get following errors at the bottom.

`Control 134217728 is not supported
Could not set Laser Power to 16
Control 134217729 is not supported
Could not set Accuracy to 2
Control 134217730 is not supported
Could not set Motion Range Trade Off to 0
Control 134217731 is not supported
Could not set Filter Option to 5
Control 134217732 is not supported
Could not set Confidence Threshold to 6
`

In other issues where this problem is discussed I couldn't help myself with the advice given there.
I tried this: https://github.com/teknotus/depthview/tree/udev
as well as this: https://github.com/PercATI/RealSense_ROS/blob/master/r200_install/ds_uvcdriver/scripts/patch_ubuntu_uvc.sh

With this (https://github.com/teknotus/depthview/tree/kernelpatchfmt) I had a few errors and wasn't sure if that works for the R200.

I'm glad for any kind of support, even small steps since I'm not that familiar with many topics.
Thank you in advance.

ddiakopoulos, 2016-04-21:
Hi @CruJon this seems to be related to the intel-ros node which uses librealsense. There could be underlying configuration errors, but this system isn't really friendly to new users who are simultaneously tackling Linux, ROS, V4L2 and new hardware. You have a lot of links from other repositories that aren't related to librealsense nor the intel-ros node. I can't really understand the problem with the info you've given. 

1. Make sure the camera works on a Windows machine using librealsense to double check the camera
2. Start with a completely fresh Ubuntu system and see if you can test only with librealsense and stream data
3. Only after a librealsense example shows streaming data, attempt to install the intel ros-node

ddiakopoulos, 2016-04-25:
Any update @CruJon ?

CruJon, 2016-04-25:
Hi ddiakopoulos, 

sorry for not getting back earlier.
Thanks a lot for your feedback.

Camera works on Windows and also with Ubuntu when I run cpp-capture.cpp
Today I tried the realsense/camera package with `roslaunch realsense_camera realsense_r200_nodelet_standalone_preset.launch` where the camera/color/image_raw picture looks good but the all camera/depth/image_raw and camera/infrared1/ topics are missing. When I select one of these topics with rqt_image_viewer, I get the following error: 
`[ERROR] [1461615926.001870006]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
Tomorrow I take some more time to get into this error, but if you have a suggestion than that would save me some time...

When I switch back to the package by BlazingForests, the errors described in my initial post remain.

Up until now I didn't delete and reinstall Ubuntu. 
I also tried different USB ports; didn't help.

Thanks for feedback!

ddiakopoulos, 2016-05-07:
Hi @CruJon with this info, I think it would make sense to file an issue on the https://github.com/intel-ros/realsense repo instead. These issues seem to be related to the ROS wrapper and not this underlying capture lib.

closed 2016-05-07 by ddiakopoulos