realsenseai/librealsense #119: ubuntu-mainline Hunk #1 FAILED

chaosink, 2016-04-23:
| Required Info  |   |
|---|---|
| Camera Model | F200 |
| Firmware Version |   |
| Operating System & Version | Linux Mint 17.2 |
| Kernel Version (Linux Only)  | 3.13.0-24-generic |
| Build System |  Makefile |

patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 FAILED at 108.
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/uvc_driver.c.rej
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #2 succeeded at 130 (offset -15 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 403 (offset -176 lines).

ddiakopoulos, 2016-04-23:
Hi @chaosink we've never tested with the 3.13 kernel on Mint and therefore cannot provide much support. Although it's not recommended, the easiest way to get up and running with a RealSense camera is through the libuvc backend.

jokla, 2016-04-25:
I have the same error in Ubuntu 14.04  (kernel 3.13.0-53-generic). If I run the example bin/cpp-capture the computer freezes and I have to restart it.

Is the patch supposed to work on Ubuntu?

Thanks!

ddiakopoulos, 2016-04-25:
@jokla the patch is suitably validated to be applied to kernel 3.19 and above only. Again, I recommend the less-stable libuvc backend for platforms we don't support.

teknotus, 2016-04-25:
So what is happening is that Hunk #1 is patching the list of rules matching
the GUID of the video format with the FOURCC name used with V4L2, and a
human friendly name. All it needs to do is add the matching rules needed
for the video formats unique to RealSense cameras. The order of that array
isn't important. Patch is looking to splice in between V4L2_PIX_FMT_Y16
and V4L2_PIX_FMT_SBGGR8. Unfortunately the human friendly name part of
V4L2_PIX_FMT_SBGGR8

    .name           = "BGGR Bayer (BY8 )",

changed in 2014 and so patch cannot find the place to splice in kernel
versions released before that change. I think an easy fix to have this
patch work on a wider range of kernel versions would be to splice the
RealSense formats into a place in the list that hasn't changed in a longer
period of time.

From "git blame" I can see that between V4L2_PIX_FMT_MJPEG
and V4L2_PIX_FMT_YVU420 hasn't changed since June 2008. At the RealSense
hackathon about a week ago someone was having the same problem on Ubuntu
14.04 and I manually edited his kernel source to fix it.

ddiakopoulos, 2016-05-14:
No movement on this issue for 3 weeks so closing it out.

closed 2016-05-14 by ddiakopoulos

robertxchen66, 2016-07-27:
I have the same problem with kernel: Linux-desktop 4.4.0-040400-generic #201601101930 SMP Mon Jan 11 00:32:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

Can you look at it again?

dianchen96, 2016-07-27:
I have the same issue with kernel: Linux version 3.13.0-92-generic (buildd@lgw01-52) (gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) ). I can successfully build the cpp files, but whenever I try to run the tutorial executable, the kernel panics. Could you please take a look at it again?

mfe7, 2016-09-16:
I also have this issue:
`patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 168 with fuzz 2 (offset 20 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 115 with fuzz 2 (offset -4 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 621.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej`

on Ubuntu 14.04 kernel 4.4.0-36-generic. However, dmesg outputs:
`[ 2773.969548] media: Linux media interface: v0.10
[ 2773.977361] Linux video capture interface: v2.00
[ 2773.988714] usbcore: registered new interface driver uvcvideo
[ 2773.988717] USB Video Class driver (1.1.1)`

petern3, 2016-11-23:
I had the same issue as mfe7, on Ubuntu 14.04 kernel 4.4.0-47-generic, and was able to solve it.

First I ran [patch-uvcvideo-ubuntu-mainline.sh](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-uvcvideo-ubuntu-mainline.sh) line-by-line - line 25 failed. So I went into [realsense-camera-formats.patch](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-camera-formats.patch).

Fortunately the patches are human-readable, and so I manually applied the patches. The files it was trying to change weren't quite what they were expecting.

After that I ran the remaining lines from patch-uvcvideo-ubuntu-mainline.sh.