realsenseai/librealsense #122: MinnowBoard Turbot can't get depth image running Liunx!

MEIXuYan, 2016-04-26:
| Required Info  |   |
|---|---|
| Camera Model | R200| 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04.4  |
| Kernel Version (Linux Only)  |  4.4.0 |
| Build System |  Makefile | 

Two month ago I tried linux but failed because of my board is no x86.
I bought a **minnowboard turbot** this time(it claim the new version of minnowboard max)
I download the ubuntu iso at  [http://wiki.minnowboard.org/Linux  ](url)
When connect to r200, I can only get color image but no depth image.
When connect to f200, it works well with "cpp-Multicam",but it show some wrong image with "cpp-capture"
When connect to sr300, "cpp-Multicam" show only color image and few seconds later it crushed.
for addition:the firmware version of my cameras：
f200    2.60.0.0
r200    1.0.72.06
sr300  3.10.10.0
[image]
[image]
[image]
[image]

JSewell-8, 2016-04-26:
Experiencing the same issue on Ubuntu using an Intel NUC. R200 camera 1.0.72.06 with only camera stream, no depth stream.

JSewell-8, 2016-04-26:
I think it might be an issue with how the image is being displayed. In my own application I am able to save the depth images to file and they look fine.

ddiakopoulos, 2016-04-26:
Hi All. These issues are 100% consistent behavior with an incorrectly patched uvcvideo.ko driver. A quick look at your `dmesg` log should also reveal messages about unknown video formats when the camera is plugged in.

It's best to run `./scripts/patch-uvcvideo-4.4.sh v4.4-wily` and copy the output of the process here. After that is complete, remember that you need to `sudo modprobe uvcvideo` to re-load the newly patched driver. Ensure that no cameras are plugged into your system while you do this.

MEIXuYan, 2016-04-26:
I have redo that step,but it do not fix.
it show "./patch-uvcvideo-4.4.sh: line 47:../script/realsense-camera-formats.patch:No such file or dictionary"
[image]

ddiakopoulos, 2016-04-26:
@MEIXuYan ok, from the output the script is clearly failing and exiting early "no such file or directory" based on this line in the script: https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-uvcvideo-4.4.sh#L47 It is stopping there and not performing the rest of the script. 

Maybe you are running the script from the wrong directory?

MEIXuYan, 2016-04-26:
Is this dictionary right in this image?

[image]

MEIXuYan, 2016-04-26:
I have tried install linux on other computer,and it worked well .
Can this problem be the oem version linux for minnowboard is not adaptive with librealsense? @ddiakopoulos

ddiakopoulos, 2016-04-26:
Yes, looks fine. I suspect you might be running `./patch-uvcvideo-4.4.sh v4.4-wily` from within the `scripts` directory instead of `./scripts/patch-uvcvideo-4.4.sh v4.4-wily` from within the root librealsense directory

ddiakopoulos, 2016-04-26:
@MEIXuYan the Minnowboard is fully compatible. It is an installation problem, not an issue with librealsense software or the hardware.

MEIXuYan, 2016-04-26:
My bad  :( !!!!!!
I have try install liunx more than three times but repeat the same mistake.
**I run ./patch-uvcvideo-4.4.sh v4.4-wily. in the dictionary of  /script** !
Hope no one will make this mistake again. @JSewell-8 
Thank you so much for your help @ddiakopoulos

closed 2016-04-26 by ddiakopoulos