realsenseai/librealsense #123: Ubuntu 16.04 support

bloyl, 2016-04-28:
Is there any plan to have Ubuntu 16.04 as an officially supported platform with instructions on how to patch the kernel?

If not, what would you suggest as a stable linux development platform? 14.04?

Thanks

mattcurfman, 2016-04-28:
Some information ...

The section "Ubuntu 14.04 LTS Installation" should be renamed "Ubuntu Installation", but there _is_ a section underneath called "Kernel in 16.04.xx" which says that no patching is required for R200 cameras.  F200 and SR300 cameras will still need a patch, and the script or documentation should be updated to include that information which is currently missing.

bloyl, 2016-04-29:
Thanks I'll give it try tomorrow.
Does any on have the script to do that patching for the F200 module? WIll a patched module work for both R200 and F200 cameras or will I need to switch modules if I change cameras?

Thanks for the help.

mkhansen-intel, 2016-04-29:
I'm seeing that when I patch the 4.4-wily kernel, F200 and SR300 show the depth and IR streams correctly (using cpp-capture) but not the color stream. Does anyone else see this problem?

bloyl, 2016-05-05:
So with no problems with the stock 16.04 kernel and an F200 camera so far.

Thanks for the help.

closed 2016-05-05 by bloyl