realsenseai/librealsense #125: Clock synchronization

jmaye, 2016-04-29:
I was just doing some tests lately comparing the timestamps of depth/IR and color frames. It seems that they are off by a couple of milliseconds (not constant). How is this handled in the RealSense hardware? Do the IR and RGB cameras have different clocks? Don't you have hardware synchronization/trigger? Are we sure that frame data gathered after a call to wait_for_frames() has been captured at the very same time?
Thanks in advance for clarifications.

ddiakopoulos, 2016-04-29:
@jmaye what cameras are you testing? Each of them have significantly different sync mechanisms.

jmaye, 2016-04-29:
I'm currently testing the F200, want to go for the SR300 next. Then, info for both would be cool.

dorodnic, 2016-04-29:
By default RGB and IR/Depth pipes on the F200 & SR300 are not synchronized. 
SR300 introduces hardware sync. This feature is currently available on Windows through RealSense SDK, but unfortunately not yet in librealsense.

jmaye, 2016-04-29:
Thanks for the reply! I'd be glad to test this hardware synchronization on librealsense, looking forward to its implementation.

ddiakopoulos, 2016-04-29:
One thing to note is that the R200 is hardware synchronized and will provide matched frame sets.

jmaye, 2016-05-02:
Great guys, thanks!

closed 2016-05-02 by jmaye