realsenseai/librealsense #126: Cannot access /sys/class/video4linux

marcelinomalmeidan, 2016-04-29:
| Required Info  |   |
|---|---|
| Camera Model | R200| 
| Firmware Version |   | 
| Operating System & Version | Ubunto 14.04 LTS  |
| Kernel Version (Linux Only)  |  4.4.4-040404-generic |

Hey everyone. I followed the instructions in the installatio.md file step by step. When trying to tun the ./cpp-capture file, the following error shows up:
Cannot access /sys/class/video4linux

Before trying to install librealsense, I was using the ROS development https://github.com/PercATI/RealSense_ROS .
I tried to go back to that one, and not even that works anymore. It doesn't recognize the camera.

Would anyone have any idea of what is going on here?

marcelinomalmeidan, 2016-04-29:
I installed it again through the whole solution described in:
https://github.com/intel-ros/realsense/issues/12

Now it worked. I don't know why it wasn't working before, I don't recall doing anything different.

closed 2016-04-29 by marcelinomalmeidan

Thomas00010111, 2016-05-23:
Had the same issue. Followed the installation process, no luck.
I think that simply plugging in the camera finally solved  this issue. Is the directory "/sys/class/video4linux" created when the camera is plugged in for the first time?

mattanimation, 2016-09-15:
I am having the same issue but using the [Intel Robotics Kit](https://01.org/developerjourney/recipe/intel-realsense-robotic-development-kit) that features the UP board. followed all the instructions, but get the same error when trying to run the point cloud examples. it seems the device is not listed when listing the USB devices, just shows a usb 3.0 hub. Other then a different Kernel for the UP board, I can't seem to figure out why the R200 isn't detected.

EdXian, 2017-11-13:
Hi @mattanimation ,
Have you resolved the problem? 
I have the same issue !!

Tianhuanyu, 2018-08-29:
i have a same question in d415.
(backend-v4l2.cpp:518) Cannot access /sys/class/video4linux

shrijan00, 2018-10-31:
> i have a same question in d415.
> (backend-v4l2.cpp:518) Cannot access /sys/class/video4linux

Were u able to fix it?>

jav-solo, 2018-12-21:
I have had a similar issue. Instead of reinstalling Ubuntu, I changed the kernel. I was previously using 4.15.0-43, and I would get this error when trying to see output from the D415.

To do this, I uninstalled all librealsense packages using:
 `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Because of an issue related to UEFI secure boot I figure out that I had to go to comment out the line:
`GRUB_HIDDEN_TIMEOUT=0` by placing a "#" in front in the file /etc/default/grub

I then rebooted in GRUB and using the advanced options for Ubuntu I changed the kernel to 4.15.0-42

I reinstalled librealsense using:
`sudo apt-get install liberealsense2-dkms` and `sudo apt-get install liberealsense2-utils`

After that my D415 worked using `realsense-viewer`