realsenseai/librealsense #127: Unable install on mac OSX

Zieng, 2016-05-01:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | 

When use commandline `make install` to install after  `make` successfully:

the result is:
```
mkdir -p obj/libuvc
mkdir -p lib
mkdir -p bin
c++ -std=c++11 -shared obj/verify.o obj/context.o obj/device.o obj/f200-private.o obj/f200.o obj/image.o obj/log.o obj/r200-private.o obj/r200.o obj/rs.o obj/stream.o obj/sync.o obj/types.o obj/uvc-libuvc.o obj/uvc-v4l2.o obj/uvc-wmf.o obj/uvc.o obj/libuvc/ctrl.o obj/libuvc/dev.o obj/libuvc/diag.o obj/libuvc/frame.o obj/libuvc/init.o obj/libuvc/stream.o `pkg-config --cflags --libs libusb-1.0` -o lib/librealsense.so
install -m755 -d /usr/local/include/librealsense
cp -r include/librealsense/* /usr/local/include/librealsense
cp lib/librealsense.so /usr/local/lib
ldconfig
make: ldconfig: No such file or directory
make: *** [install] Error 1
```
I know that `ldconfig` is a linux command, and in OSX there is a command called `dyld` has similar function.
But simply replace `ldconfig` with `dyld` won't work.

Do I miss something?

ddiakopoulos, 2016-05-01:
Please use the included XCode solution instead of the makefile. The makefile is not supported on OSX.

Zieng, 2016-05-03:
@ddiakopoulos 
tried the XCode, still not OK.
for example, first I build the `librealsense` successfully, then I connected the device (F200) via USB, tried the `cpp-tutorial-3-pointcloud`, and the result is:
```
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense F200
    Serial number: 039140069409
    Firmware version: 2.50.0.0
rs.error: TemperatureControlLoop: libusb_bulk_transfer(...) returned LIBUSB_ERROR_NO_DEVICE
rs::error was thrown when calling rs_wait_for_frames(device:0x103000018):
    Timeout waiting for frames.
Program ended with exit code: 1
```
Any suggestions?

ddiakopoulos, 2016-05-03:
@Zieng could be a hardware issue... often indicated by `Timeout Waiting for Frames`. But...you are running on an older F200 firmware that can only be updated by a Windows 10 computer by installing the RealSense F200 DCM. That could be one issue... so I would try updating first.

Zieng, 2016-05-03:
Thanks a lot!
I will try to update the firmware and check if it's OK.

Zieng, 2016-05-03:
@ddiakopoulos 
I update firmware to latest, but still not working.
here is the message:
```
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense F200
    Serial number: 039140069409
    Firmware version: 2.60.0.0
rs_error was raised when calling rs_wait_for_frames(device:0x102002018):
    Timeout waiting for frames.
Program ended with exit code: 1
```

ddiakopoulos, 2016-05-03:
@Zieng maybe try on a different USB port or through a powered USB 3.0 hub.

blackerpaper, 2016-05-13:
If you using hackintosh, it didn't work.
It work in my macbook air mid 2012. Libusb using from brew.

Zieng, 2016-05-14:
@blackerpaper
Thanks for your reply. My computer is rmbp (early 2015), OSX 10.11.4.

thiagopradi, 2016-06-09:
I'm experiencing the same issue. In my case, the RGB camera works fine, but If I try to get depth or IR frames, I get the following error:

````
RealSense error calling rs_wait_for_frames(device:0x7ffb49003618):
    Timeout waiting for frames.
````

Hardware information:
Intel RealSense F200 - Firmware version: 2.60.0.0
MacBook Pro (Retina, 13-inch, Early 2015)

blackerpaper, 2016-06-09:
I know that it seems trivial, have you tried to reset the smc or pram?

According to:
https://discussions.apple.com/thread/7076646?start=0&tstart=0

ddiakopoulos, 2016-06-25:
Seems like the root cause here is USB controller H/W that is just flaky for some versions of RealSense cameras on some versions of Apple hardware. Unfortunately there's not much we can do here other than keep a running list of Apple H/W that has known incompatibilities.

closed 2016-06-25 by ddiakopoulos

ataiya, 2016-07-05:
Regarding this comment:

> "Please use the included XCode solution instead of the makefile. The makefile is not supported on OSX"  

If that is the case, then you should have a clear error that checks for OSX and informs the user!!!

ddiakopoulos, 2016-07-05:
@ataiya Not supported is precisely what that means. Several people have contributed steps to enable Makefile support on OSX that I have merged. I've never tested them and therefore have no indication if it's working or broken. Since this is an open-source project with a large number of merged contributions, I highly support your time and effort figuring out why it's not working _on your system_ and submitting a fix!

ataiya, 2016-07-05:
Alright, nice to hear that. Give me a bit of time and I will fix the CMake build for OSX.
(XCode is just too ugly... polluted my ~/Library/Developer with build files... sad)

billingo, 2018-05-21:
I see it was quite a while ago this issue was discussed. I'm running into exactly this issue with my R200 on OSX 10.12. Any progress since this issue was discussed?