realsenseai/librealsense #131: F200 driver corrupted?

bpwiselybabu, 2016-05-05:
| Required Info  |   |
|---|---|
| Camera Model |  F200  | 
| Firmware Version |  - | 
| Operating System & Version | Linux Ubuntu 14.04  |
| Kernel Version (Linux Only)  |  4.4.6  |
| Build System |  Makefile |

Hi,

   My F200 stopped working. When I try to access it I get  

RealSense error calling rs_create_context(api_version:4):
    Cannot access /sys/class/video4linux

    dmesg | tail -n 20

[17926.454449] usb 1-14: New USB device found, idVendor=05e3, idProduct=0610
[17926.454455] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[17926.454458] usb 1-14: Product: USB2.0 Hub
[17926.454461] usb 1-14: Manufacturer: GenesysLogic
[17926.454913] hub 1-14:1.0: USB hub found
[17926.455434] hub 1-14:1.0: 2 ports detected
[17927.162465] usb 1-14.2: new full-speed USB device number 11 using xhci_hcd
[17927.407751] usb 1-14.2: New USB device found, idVendor=041e, idProduct=4099
[17927.407757] usb 1-14.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[17927.407760] usb 1-14.2: Product: Creative VF0800
[17927.407763] usb 1-14.2: Manufacturer: Creative Technology Ltd
[17927.407765] usb 1-14.2: SerialNumber: 201503231187
[17927.543692] input: Creative Technology Ltd Creative VF0800 as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14.2/1-14.2:1.3/0003:041E:4099.0006/input/input19
[17927.599064] hid-generic 0003:041E:4099.0006: input,hiddev0,hidraw3: USB HID v1.11 Device [Creative Technology Ltd Creative VF0800] on usb-0000:00:14.0-14.2/input3
[17928.867893] usb 2-4.4.1: new SuperSpeed USB device number 11 using xhci_hcd
[17928.884419] usb 2-4.4.1: New USB device found, idVendor=8086, idProduct=0a95
[17928.884424] usb 2-4.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[17928.884427] usb 2-4.4.1: Product: Intel(R) RealSense(TM) 3D Camera (Front F200) Device
[17928.884430] usb 2-4.4.1: Manufacturer: Intel \xffffffe5\xffffffa0\xffffff91\xffffffa0\xffffff91\xffffffcd\xffffffaa\xffffffaaI
[17928.884432] usb 2-4.4.1: SerialNumber: 111111111111

Is there any way to reinstall the firmware?

teknotus, 2016-05-06:
What do you get with...

lsusb -v -d 8086:0a95

I think that is the device firmware upgrade USB ID.

> Is there any way to reinstall the firmware?

Probably the same way as a firmware upgrade. From a USB log of a firmware
upgrade I can say that the implementation of the DFU protocol on the F200
is broken enough that the standard Linux tools aren't going to work. The
windows tools are supposed to automatically install new firware so
connection to a windows machine might work. If the factory calibration is
corrupted that is another matter.

bpwiselybabu, 2016-05-06:
lsusb -v -d 8086:0a95 gives

Bus 002 Device 018: ID 8086:0a95 Intel Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass           14 Video
  bDeviceSubClass         0 Undefined
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
  idProduct          0x0a95 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           62
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      1 Device Firmware Update
      bInterfaceProtocol      1 
      iInterface              2 
      Device Firmware Upgrade Interface Descriptor:
        bLength                             9
        bDescriptorType                    33
        bmAttributes                      255
          (unknown attributes!)
          Will Detach
          Manifestation Tolerant
          Upload Supported
          Download Supported
        wDetachTimeout                    513 milliseconds
        wTransferSize                    1024 bytes
        bcdDFUVersion                   2.01

I have tried to connect it to a windows 10 computer and I cant get it to do a firmware upgrade on it as it cant detect it. I have a feeling i have bricked this R200..

teknotus, 2016-05-06:
>
> I have tried to connect it to a windows 10 computer and I cant get it to
> do a firmware upgrade on it as it cant detect it. I have a feeling i have
> bricked this R200..
>
>
> Hmm. Windows must be only looking for cameras in firmware update mode
after it tells them to go into firmware update mode.

Can possibly trick it by putting the descriptor from a working F200 with an
old firmware version on a programmable USB device like a teensy which might
prompt it to start the process. and cause it to find the real camera
already in firmware update mode. I probably have the tools to test this.

Floe posted a packet capture of the firmware update process. Should be able
to extract the firmware from that, and hack the Linux dfu utils...
https://github.com/teknotus/depthview/issues/12

Can also ask on the official RealSense forum. Seems like a simple check for
cameras in update mode would be enough for the standard tools to fix your
camera.

ddiakopoulos, 2016-05-14:
@bpwiselybabu try getting a replacement camera from click.support@intel.com.

@teknotus R200 doesn't use DFU for firmware update (although F200 and SR300 do). In either case, I don't think it's possible to keep a camera in firmware update mode across power cycles, so I don't think there's an issue with that.

teknotus, 2016-05-14:
> @teknotus R200 doesn't use DFU for firmware update (although F200 and
SR300 do).

Yeah I know. They said R200 but posted an lsusb retrieved descriptor for an
F200 in DFU mode 8086:0a95. lsusb cannot get the product string without
sudo or it would have been clearer.

> In either case, I don't think it's possible to keep a camera in firmware
update mode across power cycles, so I don't think there's an issue with
that.

If if doesn't have a valid firware it would make sense to stay in DFU mode.
I'm not ready to test that idea by intentionally erasing the firmware
though.

ddiakopoulos, 2016-05-29:
@bpwiselybabu maybe you could try plugging it into a Windows machine and installing this specific version of the RealSense DCM (which initiates a firmware upgrade, something that librealsense doesn't support currently): https://github.com/IntelRealSense/librealsense/issues/151#issuecomment-221239906

dnpwebdesign, 2016-06-13:
My HP Spectre X2 in-built front camera F200 is undetectable on Fedora 23 Linux Kernel 4.5.6. The rear camera R200 is detected in cameras, namely /dev/video0, /dev/video1, /dev/video2. Works on Skype and Cheese. I have followed all the steps to install the library, it does have no effect on R200, and F200 is still not detected. Is there a patch, or setup I can do to make it work?

ddiakopoulos, 2016-06-13:
Hi @dnpwebdesign, Spectre x2 only has an R200 camera. There are no other RealSense cameras in the system.

dnpwebdesign, 2016-06-13:
Wow, I didnt know that, so what is the front camera type and brand? Isnt it intel as well?

ddiakopoulos, 2016-06-13:
@dnpwebdesign No idea! I think it's a regular webcam. No particular brand, probably the reference design.

teknotus, 2016-06-13:
The HP Spectre X2 has two MIPI CSI-2 cameras. Both a front facing, and a
back facing one. I think the i2c part of the camera interface shows up from
Linux, but the data lanes, and Image Signal Processor (ISP) don't have
support.

dnpwebdesign, 2016-06-15:
All cameras in Spectre works in Windows 10, but I guess, the art of happiness comes from Linux.
The rear camera, which was detected as R200 in Ubuntu/Debian/Fedora, can capture 3 different images: 1 flat green, 1 semi wobbly green, 1 normal 1080. 
I wonder why Linux does not support the data lanes and ISP for the front one, since hybrid devices get more popular such as the MS Surface Pro line of products

So there is really no work around this?

teknotus, 2016-06-15:
>
> All cameras in Spectre works in Windows 10, but I guess, the art of
> happiness comes from Linux.
>
There are actually 3 color cameras on the Spectre. The RealSense R200 has a
rear facing color camera with a USB interface, and there are both front,
and back MIPI CSI-2 cameras as well.
The windows 10 webcam application only uses the MIPI CSI-2 cameras.

> The rear camera, which was detected as R200 in Ubuntu/Debian/Fedora, can
> capture 3 different images: 1 flat green, 1 semi wobbly green, 1 normal
> 1080.
>
These are the Depth, IR, and color sources from the R200. Librealsense
should get you access to all of them.

> I wonder why Linux does not support the data lanes and ISP for the front
> one, since hybrid devices get more popular such as the MS Surface Pro line
> of products
>
Windows handles these cameras with 4 drivers all written by Intel. There is
a MIPI CSI-2 driver, an Intel ISP driver, and two different Omnivision
camera drivers. The MIPI CSI-2 interface, and the Intel ISP were introduced
for Atom System On a Chip (SOC) a few years ago. Intel released a kernel
patch supporting both, but seems to never have upstreamed the drivers. It's
hard to find a link to download the patch, but I'm guessing it's very
similar to what is needed with Skylake systems. I wasn't able to find any
public documentation on the interface so modifying them to work with
Skylake could be difficult.

The MIPI specifications themselves are under NDA, and only available to
companies in the MIPI alliance. It seems to cost $8k a month to join. There
are however so many open source drivers for MIPI that it's probably not
that hard to deduce enough of the specifics to interface with it.

Omnivision has all of their documentation under NDA as well so there are
similar problems there, but they probably work very similarly to other
omnivision cameras that have open source drivers.

On the bright side there seems to be at least support for the MIPI
interface already in the kernel. I think these are the MIPI interfaces on
my Spectre X2
/sys/devices/pci0000:00/0000:00:15.0
/sys/devices/pci0000:00/0000:00:15.2
/sys/devices/pci0000:00/0000:00:15.3

Only two of these would be physically connected to a camera. They each have
an i2c interface which would be the camera controls. They also have two DMA
channels which I would think are either the two data lanes, or the in, and
out buffers. There is a good chance this works the same as the patch Intel
released for the Atom. There are a bunch of CSI-2 omnivision drivers in the
kernel. Probably one is close enough to guess at how to get some data.

This leaves the ISP without support. The ISP is responsible for turning the
RAW image into the formats most camera software wants. There are several
open source libraries that can convert so you would lose performance, and
convenience.

> So there is really no work around this?
>
There are currently no ready to go solutions, but the fact that using this
interface slightly reduces the cost of making a laptop verses a USB camera
means many Skylake laptops are going to use it, and increases the demand
for someone to solve the problem. I'm not sure where to ask at Intel if
there is anything in progress. Maybe the maintainer of the driver that
exposed the MIPI CSI-2 ports.

dnpwebdesign, 2016-06-16:
Excellent review.

I'm pretty sure that should be a driver or kernel patch that can fix it.
I found 2 new interesting facts on Windows:
- There is are 3 R200 devices recognised namely L/R, Front, and Rear. 
- There is another suspicious driver called Intel AVStream 2500. I wonder if this is the required culprit we need to mimic for Linux.

I wonder if there are actually 2 cameras facing to the front 1 being R200, the other is AVStream.

teknotus, 2016-06-16:
> There is another suspicious driver called Intel AVStream 2500.

I see the number 2500 twice in device manager.  One of them says "Image
Signal Processor" in the name, and the other has a "Bus reported device
description" of "ISP Camera Device". ISP is an acronym for Image Signal
Processor.

One of them is an audio device. I suspect besides the usual raw image to
YUV it might have something fancy like H264 encoder that can handle audio
and that is why it shows up twice.

ddiakopoulos, 2016-06-25:
There's too many different topics being discussed or documented in this issue. Closing it out -- feel free to re-open new issues as they pertain to librealsense functionality or bugs.

closed 2016-06-25 by ddiakopoulos