realsenseai/librealsense #132: Cannot access /sys/class/video4linux

ckim337, 2016-05-06:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | -  | 
| Operating System & Version | Linux Ubuntu 14.04  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | 

Hi, I am currently using SR300. When I try to launch the SR300, it shows
[ERROR] [1462497825.965532722]: RealSense Camera - Error calling rs_create_context ( api_version:4 ): 
Cannot access /sys/class/video4linux
 After dmesg | tail -n 20
[   11.938464] init: catkin main process (3338) terminated with status 1
[   11.938473] init: catkin main process ended, respawning
[   11.966431] init: PR2 main process (3361) terminated with status 1
[   11.966442] init: PR2 main process ended, respawning
[   12.319642] init: kinect2 main process (3508) terminated with status 1
[   12.319653] init: kinect2 main process ended, respawning
[   12.341002] init: catkin main process (3539) terminated with status 1
[   12.341012] init: catkin main process ended, respawning
[   12.403044] init: PR2 main process (3564) terminated with status 1
[   12.403056] init: PR2 main process ended, respawning
[   12.719484] init: kinect2 main process (3710) terminated with status 1
[   12.719495] init: kinect2 respawning too fast, stopped
[   12.756007] init: catkin main process (3727) terminated with status 1
[   12.756014] init: catkin respawning too fast, stopped
[   12.800666] init: PR2 main process (3767) terminated with status 1
[   12.800673] init: PR2 respawning too fast, stopped
[   33.577037] audit_printk_skb: 207 callbacks suppressed
[   33.577039] audit: type=1400 audit(1462497451.449:80): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=4927 comm="apparmor_parser"
[   33.577045] audit: type=1400 audit(1462497451.449:81): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=4927 comm="apparmor_parser"
[   33.577276] audit: type=1400 audit(1462497451.449:82): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=4927 comm="apparmor_parser"

Is there anyway that I can solve this problem?

ddiakopoulos, 2016-05-06:
Does the SR300 work with our example applications? A cannot access v4l2 error is usually related to a missing or mis-configured udev rule (consult installation.md file for reference on that). 3.16 kernel has never been tested on our end.

ckim337, 2016-05-06:
@ddiakopoulos 
Even with the Window 10, it is not working with SDK sample browser.
under "device manager-imaging devices", I see both SR300 Depth and RGB. However, in SDK sample browser-Tools-camera explorer, it says not connected in both (front) and (long range).

It worked for a day in Linux ubuntu 14.04, then stopped working with showing that error. 
Do you have any idea ? :(

ddiakopoulos, 2016-05-14:
Hi @ckim337 - It seems like you might be describing a hardware issue if the camera doesn't work on either Windows 10 with the official SDK nor Ubuntu 14.04 with librealsense. I would recommend attempting to RMA the unit by emailing click.support@intel.com. For the past few weeks, SR300 has been out of stock, so expect a delay. I will close this Github issue since don't think it's a software problem -- but please re-open if you get a new camera and it's still not working.

closed 2016-05-14 by ddiakopoulos

ille-90, 2016-10-10:
Had the same problem with the R200:
Followed the guide [developerjourney](https://01.org/developerjourney/recipe/intel-realsense-robotic-development-kit) until step 7.

Command `bin/cpp-pointcloud` gave me 
>RealSense error calling rs_create_context(api_version:4):
    Cannot access /sys/class/video4linux
Command `dmesg | tail -n 20` gave me 
>[ 1403.235452] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1456.380974] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1480.000896] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1486.059762] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1492.126740] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1504.036349] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1527.651360] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1557.102281] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1563.168907] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1592.007320] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1661.930588] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1670.348370] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1674.497774] systemd-hostnamed[4355]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[ 1705.457720] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1729.785619] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1735.844008] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1741.291367] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1777.316070] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1782.763168] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[ 1795.392689] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?

So i changed the USB 3.0 Adapter+ USB 3.0 cable to the single micro cable and it worked instant :)