realsenseai/librealsense #133: Problem with driver

liborw, 2016-05-06:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version |  Linux arch 4.5.1-1-ARCH  |
| Kernel Version (Linux Only)  | 4.5.1-1-ARCH  |
| Build System |  Makefile | 

I'm able to view the streams using mpv but the examples don't work:

First try:
```
$ ./cpp-capture 
Capturing DEPTH at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
Capturing COLOR at 640 x 480, fov = 55.3 x 42.5, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
Segmentation fault (core dumped)
```
Second:
```
$ ./cpp-capture 
Capturing DEPTH at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
Capturing COLOR at 640 x 480, fov = 55.3 x 42.5, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
RealSense error calling rs_start_device(device:0x6a2c30):
    uvc_set_ctrl(...) returned LIBUSB_ERROR_PIPE
```

Replug
```
$ ./cpp-config-ui 
RealSense error calling rs_device_supports_option(device:0x1c8b800, option:56):
    bad enum value for argument "option"
```

```
$ ./cpp-pointcloud 
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x6599829c
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x35c735c7
RealSense error calling rs_wait_for_frames(device:0xa75120):
    Timeout waiting for frames.
```

ddiakopoulos, 2016-05-14:
Hi @liborw. The streams with MPV are a bit of a false indication that the uvcvideo.ko patch was installed correctly because most RealSense cameras will stream from a default configuration. Are you using the libuvc backend (I see LIBUSB_ERROR_PIPE above...)? In any case, we've never really tested or validated on Arch or that kernel version, so there could be a couple issues going on here... especially since the error messages aren't consistent across runs of the application.

ddiakopoulos, 2016-05-20:
@liborw any progress on fixing your issue?

liborw, 2016-05-23:
Yeah, I have new machine, now I'm able to see the color stream, but the rest is blank. The tutorial-1-depth seems to work, but for the rest I get just the:
```
rs.warn: Subdevice 0 bad magic number 0x3d003e
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x470049
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x40003e
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x4c0046
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x4a004a
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x490043
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x3d0043
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x430047
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x40003f
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x4f0047
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x45003f
rs.warn: Subdevice 1 bad magic number 0x0
```

ddiakopoulos, 2016-05-29:
@liborw Your new symptoms are still consistent with an incorrectly patched `uvcvideo.ko` driver. Next steps would be to closely monitor what the patching script is doing if it exits early or throws an error...

liborw, 2016-05-30:
I thought that for the newer kernel the patch is not necessary, i have 4.5.4.1.

liborw, 2016-06-14:
@ddiakopoulos so, I have manually patched the kernel (4.6.2-1-ARCH) uvc module, using the patch in the scripts directory. However, when I run the ./cpp-capture, i get RGB stream, but the rest freezes on the first frame. The output is:
```
$ ./cpp-capture 
Capturing DEPTH at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
Capturing COLOR at 640 x 480, fov = 55.3 x 42.5, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
...
```

imdoes, 2016-06-17:
I'm on Arch with the linux-lts kernel (4.4.13), and a quick test gives me all 3 streams functioning on an SR300. But I did have to patch & compile the kernel module. Before that I only got RGB, and the point cloud example gave what could at best be described by a few white lines running in the vertical direction.

Since I did have a computer running your kernel as well, I tried to replicate the issue (but note, with the SR300. It might be different).
Using 4.6.2-1-ARCH unpatched I couldn't get anything other than RGB as well, but with the patch it works. So apparently the patch is still required. 
Are you sure that your kernel is loading the new module? I placed mine in `/usr/lib/modules/4.6.2-1-ARCH/updates` and then ran `depmod`, followed by `modinfo uvcvideo` to see that it did in fact update the path to the module. An `rmmod` and `modprobe uvcvideo` video later and I got all 3 streams.

liborw, 2016-06-22:
I have also place the module into updates, checked that the path is updated. Tested other binaries and if I start just depth and RGB in `cpp-capture-ui` I get both streams, with just some:

```
 ./cpp-config-ui 
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
```

but when I start infrared, the app sometimes crashes with:

```
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
RealSense error calling rs_wait_for_frames(device:0x117f600):
    Timeout waiting for frames.
```

or the depth and both infrared streams doesn't stream and are stuck on the first frame.

gabrielarpino, 2016-07-05:
Hi imdoes,

I am getting the same issue. What do you mean by "i placed mine in /usr/lib/modules/4.6.2-1-ARCH/updates"? What did you place there and how? Thanks for the help :) I am still getting Subdevice 1 bad magic number 0x0 on kernel 4.2.0

jlgarcia75, 2017-09-29:
Do you still need help with this issue?

For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Best regards,
Jesus
Intel Customer Support

gabrielarpino, 2017-10-03:
No, it has been resolved, thanks!

closed 2017-10-03 by jlgarcia75