realsenseai/librealsense #136: SR300 partially working

roberto-martinmartin, 2016-05-10:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.10.10.0  | 
| Operating System & Version | Linux Ubuntu 14.04 LTS |
| Kernel Version (Linux Only)  | 4.4.8-040408-generic  |
| Build System |  Makefile | 

I followed the instructions to install librealsense. 
Then I executed cpp-capture. I can see the color images, but the depth and infrared images are strange/corrupted. The leds on the camera are on. The IR projector is on.
The outcome of sudo dmesg | tail -n 50 is:

[16317.329088] media: Linux media interface: v0.10
[16317.337447] Linux video capture interface: v2.00
[16317.352510] usbcore: registered new interface driver uvcvideo
[16317.352512] USB Video Class driver (1.1.1)
[16345.846748] usb 9-2: new high-speed USB device number 6 using xhci_hcd
[16345.870991] usb 10-2: new SuperSpeed USB device number 6 using xhci_hcd
[16345.891709] usb 10-2: New USB device found, idVendor=05e3, idProduct=0616
[16345.891714] usb 10-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[16345.891717] usb 10-2: Product: USB3.0 Hub
[16345.891719] usb 10-2: Manufacturer: GenesysLogic
[16345.892988] hub 10-2:1.0: USB hub found
[16345.893459] hub 10-2:1.0: 2 ports detected
[16345.981638] usb 9-2: New USB device found, idVendor=05e3, idProduct=0610
[16345.981643] usb 9-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[16345.981646] usb 9-2: Product: USB2.0 Hub
[16345.981649] usb 9-2: Manufacturer: GenesysLogic
[16345.982893] hub 9-2:1.0: USB hub found
[16345.983761] hub 9-2:1.0: 2 ports detected
[16346.802664] usb 9-2.2: new full-speed USB device number 7 using xhci_hcd
[16347.057011] usb 9-2.2: New USB device found, idVendor=041e, idProduct=4099
[16347.057015] usb 9-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16347.057019] usb 9-2.2: Product: Creative VF0800
[16347.057021] usb 9-2.2: Manufacturer: Creative Technology Ltd
[16347.057024] usb 9-2.2: SerialNumber: 000000032117
[16347.215608] input: Creative Technology Ltd Creative VF0800 as /devices/pci0000:00/0000:00:01.0/0000:01:00.0/usb9/9-2/9-2.2/9-2.2:1.3/0003:041E:4099.0007/input/input24
[16347.270971] hid-generic 0003:041E:4099.0007: input,hiddev0,hidraw3: USB HID v1.11 Device [Creative Technology Ltd Creative VF0800] on usb-0000:01:00.0-2.2/input3
[16348.370897] usb 10-2.1: new SuperSpeed USB device number 7 using xhci_hcd
[16348.390113] usb 10-2.1: New USB device found, idVendor=8086, idProduct=0aa5
[16348.390117] usb 10-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[16348.390121] usb 10-2.1: Product: Intel(R) RealSense(TM) Camera SR300
[16348.390123] usb 10-2.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[16348.391895] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[16348.394076] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:01.0/0000:01:00.0/usb10/10-2/10-2.1/10-2.1:1.0/input/input25
[16348.395034] uvcvideo: Unknown video format 5a564e49-2d90-4a58-920b-773f1f2c556b
[16348.395038] uvcvideo: Unknown video format 495a4e49-1a66-a242-9065-d01814a8ef8a
[16348.395040] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[16348.395042] uvcvideo: Unknown video format 49564e49-57db-5e49-8e3f-f479532b946f
[16348.395372] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[16348.397255] uvcvideo: Unable to create debugfs 10-7 directory.
[16348.397450] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:01.0/0000:01:00.0/usb10/10-2/10-2.1/10-2.1:1.2/input/input26

The sensor was working on another computer.

Do you have an idea of the problem?

Thanks

--roberto

teknotus, 2016-05-10:
uvcvideo: Unknown video format

Is an indicator of an unpatched kernel.

ddiakopoulos, 2016-05-10:
@roberto-martinmartin Yes, your log indicates that the kernel wasn't patched. Try re-running the patch script for your configuration and trying again, making note during the process if the script exits early or fails. Remember at the end of the process you will need to `sudo modprobe uvcvideo`.

roberto-martinmartin, 2016-05-10:
Ok, I will try repatching. I tried patching twice and I couldn't see any error message. I ran sudo modprobe uvcvideo after it and dmesg didn't show any weird thing. The error messages I copied appear when I connect the sensor.

I will try again and check the outcome!

roberto-martinmartin, 2016-05-10:
Ok, now it seems to work.

The problem is that somehow in the process of updating the ubuntu 14.04 kernel to 4.4 I ended up having the 4.4.8. (the update process was not that easy because the new kernel was incompatible with my nvidia driver, my unity desktop stopped working and I had to download a newer driver through the command line and install it. In this process it seems that I also updated the kernel to 4.4.8?)

The patching stopped in the step 
cp $CONFIG_LOCATION/.config .
cp $CONFIG_LOCATION/Module.symvers .

because in my case, the definition 

CONFIG_LOCATION=/usr/src/linux-headers-$RAW_TAG*-generic/

ended up having 2 locations, one for the headers of 4.4 and one for 4.4.8.

From there nothing executed.

Solution:
- I first tried executing the rest of the script manually, using as CONFIG_LOCATION the smaller number, as recommended in other issues (#12). However, this produced a module .ko that was not able to load because it was generated for kernel 4.4 and I'm running 4.4.8
- The second option was to run manually the script but defining LINUX_BRANCH=4.4.8-wily. This generated a module that loads. Probably it would not be necessary to manually execute each line of the patching script, but just call it with the argument "4.4.8-wily" instead of "4.4-wily"

I will do some tests tomorrow to be sure everything works fine.

ddiakopoulos, 2016-05-14:
@roberto-martinmartin glad you got it working! Closing this out and changing the title to support better searchability about the issues you encountered.

closed 2016-05-14 by ddiakopoulos