realsenseai/librealsense #138: different questions on multi-camera

Thanh-Binh, 2016-05-11:
I have some questions by using multi-camera:
1. How can we synchronize all cameras?
2. how is the video frame rate if all camera connect to the same USB3-port / to different USB3-port?
3. How is the interference between cameras if they are oriented to the same object?

It is very nice to hear any answer from your experience!
Best thanks

mhkabir, 2016-05-11:
Same questions! :)

dorodnic, 2016-05-11:
1. Depends on what you mean by synchronization. Temporal sync (making sure all cameras capture frames at the same time) is problematic since each camera has its distinct clock. For spacial sync, you could match objects from one camera in frames of another through image registration or estimate fixed relative camera models. Both techniques are not trivial, and are beyond the scope of librealsense right now. 
2. Using the same USB3 controller for more then one realsense device is not recommended. Bandwidth of USB3 is limited and you will inevitably loose frames. If you're using separate USB controllers, you will still need a powerful PC or you will have frame drops at the kernel driver level. 
3. The interference depends on camera type. In short, you can capture depth from two R200 devices pointed at the same target, but not from two SR300/F200. You can read more about that under [Camera Specs](https://github.com/IntelRealSense/librealsense/blob/master/doc/camera_specs.md)

Perhaps @ddiakopoulos will be able provide additional information.

Thanh-Binh, 2016-05-11:
I mean temporal sync for object tracking and 3D pose estimation.
Best thanks

bloyl, 2016-05-12:
@dorodnic 
regarding interference from multiple SR300/F200 cameras aimed at the same target do you know if anyone has used the [shake and sense](http://research.microsoft.com/pubs/171706/shake%27n%27Sense.pdf) method to alleviate this problem? Is there any reason to think that it would work less well on the SR300 and F200 then on the kinect V1?

ddiakopoulos, 2016-05-12:
@bloyl  Kinect V1 uses very traditional structured light (projected dot pattern). The SR300 and F200 both use a "coded light" approach that computes depth over a series of 11 grey-coded frames -- this means it is very sensitive to motion.

bloyl, 2016-05-12:
Thanks, that makes sense.

Are there any issues with mixing R200 cameras with an SR300 or F200? does the R200's laser cause an issue to the "coded light" depth sensing. 

How does the depth resolution of the R200 camera, used in the 0.5m-1m range match up to the F200's depth resolution?

also unrelatedly, is there a better place to ask these questions then on the bug tracker?

teknotus, 2016-05-12:
> Are there any issues with mixing R200 cameras with an SR300 or F200? does
the R200's laser cause an issue to the "coded light" depth sensing.

Actually the opposite. If an F200 is pointing at the same spot as an R200
the depth data from the F200 is mostly intact, but the R200 data is pretty
much ruined.

Thanh-Binh, 2016-05-13:
@dorodnic 
regarding the temporal synchronize between many cameras:
Is it possible to trigger camera from SW, because each camera is global shutter?

ddiakopoulos, 2016-05-14:
Hi @Thanh-Binh unfortunately not. Current-generation RealSense cameras are not designed to compete with the feature set of industrial vision sensors.

digitologist, 2016-05-14:
Have you checked the infrared viewer on the r200 when this happens?  Likely simple oversaturation and reducing the IR gain or exposure would bring it back.

closed 2016-05-20 by ddiakopoulos

RocketSumit, 2018-07-23:
I am using one USB3 controller for 4 SR300 cameras and losing frames for RGB images too.!