realsenseai/librealsense #139: Cannot get depth

bage613, 2016-05-16:
I'm using Realsense R200 and F200 Camera, which I don't know the Firmware Version(Maybe you can tell me how to get it, thanks). OS is Ubuntu 14.04 LTS, kernel version is 4.4.0. 
I install librealsense following the guide https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md. 
My notebook is Thinkpad E550, which has a pre-installed Realsense F200, and I plugged a R200.
When I execute cpp-multicam, F200 can get both color and depth, but R200 cannot get depth, error info is below:
rs.warn: Subdevice 1 bad magic number 0x0
Here is the screenshot: 
[image]
What's worse, after reboot, when I run example executables in ./bin, it will fail, and error info is below:
RealSense error calling rs_create_context(api_version:4):
    Cannot access /sys/class/video4linux

I have to execute the command: sudo modprobe uvcvideo
to make camera to work again.
Can you help me to solve the problems? Thank you!
Hope to get your reply ASAP. ^_^

BR,
Xingyi

ddiakopoulos, 2016-05-16:
Hi @bage613 These problems (no R200 depth, bad magic number) can be explained by the uvcvideo module not being patched correctly. Try re-running the patch script for your configuration and trying again, making note during the process if the script exits early or fails.

bage613, 2016-05-18:
Hi @ddiakopoulos , after tried several times, I did it successfully! There are errors such as "Failed to fetch http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu/dists/trusty/main/binary-" when running install_dependencies-4.4.sh, so I commented  "sudo add-apt-repository ppa:ubuntu-toolchain-r/test", and it worked. Thank you! I guess maybe the firewall made it failed. Here is China, you know it. Thank you again. ^_^

ddiakopoulos, 2016-05-18:
Great! Glad it worked out :)

closed 2016-05-18 by ddiakopoulos

l3s777, 2016-08-10:
hey guys! 
I still have some issues with the image in depth, it is very noisy