realsenseai/librealsense #146: Installation Problems on Ubuntu 14.04 LTS with kernel wily-4.4.4

MrGeva, 2016-05-22:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04 LTS   |
| Kernel Version (Linux Only)  |  wily-4.4.4 |
| Build System |  Makefile | 

I followed the installation instructions but this step seems to fail:
./scripts/patch-uvcvideo-4.4.sh v4.4-wily

because I see printed to console:
```
Hunk #1 succeeded at 109 (offset 1 line).
patching file drivers/media/usb/uvc/uvcvideo.h
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 594 (offset 15 lines).
cp: will not overwrite just-created ‘./.config’ with ‘/usr/src/linux-headers-4.4.4-040404-generic/.config’
```
Then when I do the next step:
`sudo modprobe uvcvideo`

I get this error:
`modprobe: ERROR: could not insert 'uvcvideo': Unknown symbol in module, or unknown parameter (see dmesg)`

ddiakopoulos, 2016-05-23:
Hello @MrGeva - We haven't tracked down the issue where this, very rarely, can occur. It has been previously documented in #70. Luckly, there is a fix available in this comment: https://github.com/IntelRealSense/librealsense/issues/70#issuecomment-200059597 and should enumerate the steps needed for a workaround.

joefriel, 2016-12-09:
I think this issue occurs because $CONFIG_LOCATION in the patching script (patch-uvcvideo-4.4.sh) contains multiple file descriptors.  This can happen if there is already an existing 4.4 kernel (not v4.4-wily) with existing headers in /usr/src/.  

$CONFIG_LOCATION is assigned using a wildcard character (*), which makes the assignment look in all folders in /usr/src.  If there are multiple matches, then $CONFIG_LOCATION contains both file descriptors separated by a space.  Then, when the copy command (cp) is executed there are three arguments (instead of the expected two) and the copy command will not allow the same filename to be copied twice in one command.

Whether it works when you manually copy depends upon which file you copy.  Using the error that is generated would lead you to copy from the second argument (instead of the first), but this might not be correct.

This could be fixed by alerting the user that there is an existing 4.4 kernel and aborting the script (although this would have the side effect of not allowing repetitive usage of the script, which might be irritating if there are other script errors that need to be debugged).  This could also be fixed by being more explicit about the version to be copied (instead of using a wildcard).  There are probably other methods that a skilled programmer might use that would be preferable.

In my case, the manual copy method did not work (as I was copying from the wrong header directory).  To make it work, I simply renamed the unwanted linux header directory so that it did not match the wildcard and then the script ran and the camera functions properly.

ev-mp, 2017-08-20:
The patch "./scripts/patch-uvcvideo-4.4.sh v4.4-wily" not relevant as it has been superseded with a Ubuntu 14/16 unified patch since [Feb' 17](https://github.com/IntelRealSense/librealsense/commit/0cb9b93f93f20b2b0ab96863badccf5fb5440177).
In the new patch the issue of referring to multiple targets while copying is resolved

closed 2017-08-20 by ev-mp