realsenseai/librealsense #147: add Python bindings

toinsson, 2016-05-22:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version | N.A | 
| Operating System & Version | All |
| Kernel Version (Linux Only)  | N.A |
| Build System | Makefile | 

Is there any plan to create a Python binding for this library ?

For memo: there are some simple bindings for the (older) Softkinetic based camera, [Snkz/DepthSense-SimpleCV](https://github.com/Snkz/DepthSense-SimpleCV) refactored/extended with this fork [pydepthsense](https://github.com/toinsson/pydepthsense).

ddiakopoulos, 2016-05-23:
Hi @toinsson. There's several internal (Intel) users who have created Python bindings. The problem here is that no one on the core dev team uses Python on a daily basis, so it would be hard to maintain as the API shifts around.

toinsson, 2016-05-24:
Ok, that's understandable : ) 
If they are willing to open source their software that could be interesting to have a look at it.

In the meantime I came up with this simple C extension: https://github.com/toinsson/pyrealsense
Might be of interest for some users of librealsense.

ddiakopoulos, 2016-05-29:
Thanks for your work on this! Closing out this particular issue, but will update if/when we ever release and maintain a Python wrapper.

toinsson, 2016-05-30:
No worries, I need this for my work : ).

leonidk, 2016-10-06:
I'm reopening this since we probably should  have and support an idiomatic [ctypes-based](https://docs.python.org/3.5/library/ctypes.html) wrapper around include/librealsense/rs.h It'd be a 1-1 mapping and wouldn't be hard to test or maintain.

toinsson, 2016-12-13:
I have updated my code, which is parsing the h files and uses ctypes now. Maybe closer to what you had in mind.

leonidk, 2016-12-19:
@toinsson that's awesome thanks

closed 2017-08-21 by dorodnic

sukumar45, 2017-09-05:
@toinsson : Do you have any example video of online usage with multiple realsense devices?