realsenseai/librealsense #148: R200: No visual effects change in 'cpp-config-ui' when adjust some options

wangji1, 2016-05-23:
| Required Info  |   | 
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version | Ubuntu 14.04.4 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System |  Makefile | 

Use 'cpp-config-ui' to test options of R200, with Start Capture, when adjust option 'COLOR_GAIN', or 'COLOR_HUE', or 'COLOR_SHARPNESS', the value be changed correctly, but there's no visual effects change. 
What's the correct behavior?

dorodnic, 2016-05-23:
For the R200 camera, COLOR_GAIN has no visual effect when COLOR_AUTO_EXPOSURE is set to 1. 
Visual effects of HUE and SHARPNESS are subtle, but clearly visible when you zoom in to the pixel level.

wangji1, 2016-05-23:
COLOR_EXPOSURE or COLOR_ENABLE_AUTO_EXPOSURE?
I can't see these 2 options in cpp-config-ui, and the range of them are 0 - 0, why it be set to 1?

ddiakopoulos, 2016-05-23:
@wangji1 This is a known bug and documented in #19 as an outstanding issue

closed 2016-05-23 by ddiakopoulos