realsenseai/librealsense #149: R200: cpp-config-ui crash when adjust R200 AUTO EXPOSURE options

wangji1, 2016-05-23:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06 | 
| Operating System & Version | Ubuntu 14.04.4 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System |  Makefile | 

There are 9 options of R200 AUTO EXPOSURE, like R200_AUTO_EXPOSURE_LEFT_EDGE; 
In cpp-config-ui, when adjust them (no matter start or stop capture), the tool will be crashed.
```
RealSense error calling rs_set_device_option(device:0x1e27d58, option:R200_AUTO_EXPOSURE_LEFT_EDGE, value:298):
    UVCIOC_CTRL_QUERY:UVC_SET_CUR error 5, Input/output error
```
```
RealSense error calling rs_set_device_option(device:0xb2bd48, option:R200_AUTO_EXPOSURE_RIGHT_EDGE, value:65237):
    UVCIOC_CTRL_QUERY:UVC_SET_CUR error 5, Input/output error
```

l3s777, 2016-08-08:
I have the same issue while testing the different values of adjustments!

dorodnic, 2016-09-06:
Please give it a try now, should be resolved. Please note that R200_AUTO_EXPOSURE_LEFT_EDGE (and other similar options are only available when R200_LR_AUTO_EXPOSURE is on.

closed 2016-09-06 by dorodnic

wangji1, 2016-09-07:
Yes, I have tried. cpp-capture does not crash now.