realsenseai/librealsense #150: R200: cpp-config-ui crash when set R200_DEPTH_CLAMP_MIN and R200_DEPTH_CLAMP_MAX

wangji1, 2016-05-23:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version | Ubuntu 14.04.4 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System | Makefile | 

In cpp-config-ui, before start capture, I could set R200_DEPTH_CLAMP_MIN and R200_DEPTH_CLAMP_MAX values, but when I set the R200_DEPTH_CLAMP_MIN value higher than R200_DEPTH_CLAMP_MAX, or set R200_DEPTH_CLAMP_MAX value lower than R200_DEPTH_CLAMP_MIN, the tool will be crashed.

It makes sense that R200_DEPTH_CLAMP_MAX value never be lower than R200_DEPTH_CLAMP_MIN, and R200_DEPTH_CLAMP_MIN value never be higher than R200_DEPTH_CLAMP_MAX, but no crash should be.

dorodnic, 2016-05-23:
config.ui is sample code, not a product. If you have different expectations you can use it as a baseline for you personal projects. 
All librealsense demos exit on first encountered error, this is the expected behavior.

closed 2016-05-23 by dorodnic

wangji1, 2016-05-23:
Thank you for quickly response.