realsenseai/librealsense #151: SR300 RGB stream disappeared

jmaye, 2016-05-23:
Hi,

The SR300 was perfectly working under Ubuntu until a colleague of mine tried it on a Windows machine with Intel SDK. There, it did not work properly. Worst of it, I cannot get back RGB stream on Ubuntu anymore, just IR and depth. Is the SR300 not compatible with Windows SDK? What have I missed?

Best

ddiakopoulos, 2016-05-23:
@jmaye The SDK has a history of nuking cameras when it does a firmware update during the DCM installation. Can you read out the firmware version of the SR300?

jmaye, 2016-05-23:
Thanks for your reply. The firmware reads: 3.15.0.0, from the call get_firmware_version. I guess that's what happened indeed. He probably hit the button for firmware update unintentionally. Is there a way to get it back running?

jmaye, 2016-05-24:
I tried downloading an older DCM for SR300, but the link is dead.

jmaye, 2016-05-24:
Problem solved by de-installing current version and installing this:
http://fichiers.touslesdrivers.com/48037/intel_rs_dcm_sr300_3.0.24.59748.exe

ddiakopoulos, 2016-05-29:
Glad you could fix it! We're still working out what to do about the firmware update situation internally.

closed 2016-05-29 by ddiakopoulos

mastrcheif, 2017-06-04:
Hi, also tried downgrading my sr300 with this version. But it first crashed during the first process of installing it and now I get on windows the error, that it needs more power to run the realsense camera.
Is there a way to restore the firmware, since I think something happened during the downgradig process.

lalalaring, 2017-10-19:
cant install this version firmware