realsenseai/librealsense #153: What is the equivalent of Device.ResetProperties?

avivhu, 2016-05-24:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | 

How do I reset the camera's properties to factory defaults?

Is there an equivalent to the Windows DCM's 
`realSenseManager.captureManager.device.ResetProperties(PXCMCapture.StreamType.STREAM_TYPE_ANY);
`
Thank you!!

dorodnic, 2016-05-29:
At the moment, librealsense does not expose default values for options. 
We are working on it, and it is likely to be available in next librealsense release.

avivhu, 2016-05-29:
All right.
Thank you.

dorodnic, 2016-09-06:
Default values can now be queried using get_option_range API.

closed 2016-09-06 by dorodnic