realsenseai/librealsense #155: Erroneous timestamps when capturing two streams with different framerates

jonathanasdf, 2016-05-27:
R200 on ubuntu g++

```
#include <chrono>
#include <iostream>
#include <librealsense/rs.hpp>
using namespace std;
using namespace std::chrono;
int main() {
  rs::context ctx;
  rs::device *device = ctx.get_device(0);
  device->enable_stream(rs::stream::color, rs::preset::highest_framerate); // 60fps
  device->enable_stream(rs::stream::depth, rs::preset::highest_framerate); // 90fps
  device->start();
  auto last = high_resolution_clock::now();
  while(true) {
    device->wait_for_frames();
    auto now = high_resolution_clock::now();
    auto color = device->get_frame_timestamp(rs::stream::color);
    auto depth = device->get_frame_timestamp(rs::stream::depth);
    cout << duration_cast<milliseconds>(now-last).count() << " "
         << color << " " << depth << " " << endl;
    last = now;
  }
  return 0;
}
```

Expected output:
the difference in timestamps should correspond to the elapsed time.

Actual output:
```
...
16 100 100 
16 111 111 
16 122 122 
16 133 133 
16 144 144 
16 155 155 
16 166 166 
16 177 177 
16 188 188 
16 200 200
...
```

ddiakopoulos, 2016-05-27:
What camera is this?

ddiakopoulos, 2016-05-28:
Indeed a bug with the R200 timestamp generation of code for specific ratios of stream framerates. There's additional timing data part of each frame we will need to pull out (the 'dinghy' row of data) to fix  it. Soon!

dorodnic, 2016-09-06:
Resolved. Please give it a try.
Feel free to re-open the issue if encounter it again

closed 2016-09-06 by dorodnic