realsenseai/librealsense #156: RealSense Camera Calibration

ktak199, 2016-05-28:
I'm working with the RealSense F200 camera calibration.
I calibrated the camera with color image and circle patterns.
Then, I transformed point cloud to circle pattern coordinate based on the code below.
But there are some errors.
What is the cause of the error ?

Camera Calibration:
[image]
Point Cloud:
[image]

`

			// Retrieve our images
			const uint16_t * depth_image = (const uint16_t *)dev->get_frame_data(rs::stream::depth);
			const uint8_t  * color_image = (const uint8_t  *)dev->get_frame_data(rs::stream::color);

			// Retrieve camera parameters for mapping between depth and color
			rs::intrinsics depth_intrin = dev->get_stream_intrinsics(rs::stream::depth);
			rs::extrinsics depth_to_color = dev->get_extrinsics(rs::stream::depth, rs::stream::color);
			rs::intrinsics color_intrin = dev->get_stream_intrinsics(rs::stream::color);
			float scale = dev->get_depth_scale();

			for (int dy = 0; dy < depth_intrin.height; ++dy)
			{
				for (int dx = 0; dx < depth_intrin.width; ++dx)
				{
					// grab our index. this will leave a small number of stragglers untouched
					int i = dy * depth_intrin.width + dx;

					// Retrieve the 16-bit depth value and map it into a depth in meters
					uint16_t depth_value = depth_image[i];
					float depth_in_meters = depth_value * scale;

					// Map from pixel coordinates in the depth image to real world co-ordinates
					rs::float2 depth_pixel = { (float)dx, (float)dy };
					rs::float3 depth_point = depth_intrin.deproject(depth_pixel, depth_in_meters);
					rs::float3 color_point = depth_to_color.transform(depth_point);
					rs::float2 color_pixel = color_intrin.project(color_point);

					pcl::PointXYZ point;
					float x, y, z;
					
					//translationMat and rotationMat are calculated with camera calibration
					
					//translation
					x = color_point.x - translationMat.at<double>(0, 0);
					y = color_point.y - translationMat.at<double>(1, 0);
					z = color_point.z - translationMat.at<double>(2, 0);

					//rotation
					point.x = x * rotationMat.at<double>(0, 0) + y * rotationMat.at<double>(0, 1) + z * rotationMat.at<double>(0, 2);
					point.y = x * rotationMat.at<double>(1, 0) + y * rotationMat.at<double>(1, 1) + z * rotationMat.at<double>(1, 2);
					point.z = x * rotationMat.at<double>(2, 0) + y * rotationMat.at<double>(2, 1) + z * rotationMat.at<double>(2, 2);`

| Required Info  |   |
|---|---|
| Camera Model |F200 | 
| Firmware Version | 2.50.0.0 | 
| Operating System & Version | Windows8.1 |
| Kernel Version (Linux Only)  |---|
| Build System |  VS2013 |

ddiakopoulos, 2016-05-28:
Hi @ktak199 - support on the topic of manual camera calibration is beyond the scope of this repository/issue tracker. I'm particularly confused why you are applying your own translation and rotation mat.

ktak199, 2016-05-28:
>support on the topic of manual camera calibration is beyond the scope of this repository/issue tracker.
My question is whether rs::intrinsics and rs::extrinsics have some error and I need manual calibration or not.

>I'm particularly confused why you are applying your own translation and rotation mat.
I want to join point clouds from multiple cameras.

Wollimayer, 2016-05-30:
@ktak199 You do apply translation befor the rotation. I think it has to be the other way around.

teknotus, 2016-06-01:
I might be missing it but I don't see distortion correction in the code. It
looks like maybe you are calibrating with the color camera instead on the
infrared camera. Also note the math properties of the standard output of
OpenCV calibration tools is different than how the factory calibration is
stored on the camera. You might need to invert the matrices.

ktak199, 2016-06-02:
@Wollimayer Thank you for your advice. I'll try it.

@teknotus Thank you for your advice.

> I might be missing it but I don't see distortion correction in the code.
According to https://github.com/IntelRealSense/librealsense/blob/master/doc/projection.md,
F200 color image have no distortion.

>  It looks like maybe you are calibrating with the color camera instead on the
infrared camera.
I am calibration with the color camera now.
I tried the infrared image, but it is hard to recognize the calibration points from the infrared image.

>  Also note the math properties of the standard output of
OpenCV calibration tools is different than how the factory calibration is
stored on the camera. You might need to invert the matrices
I omit the code. But, I invert rotationMat in the code.

ktak199, 2016-06-05:
@teknotus 
`It looks like maybe you are calibrating with the color camera instead on the infrared camera.`
I tried the infrared image, but the error is not corrected.

closed 2016-06-25 by ddiakopoulos