realsenseai/librealsense #157: execute  "make BACKEND=LIBUVC" commond failed!

wllw210, 2016-05-30:
Hi，
When I do the step "LibUVC backend" of Ubuntu 14.04 LTS Installation, got the errors as below:
mkdir -p obj/libuvc
mkdir -p lib
mkdir -p bin
g++ -std=c++11 -shared obj/verify.o obj/context.o obj/device.o obj/f200.o obj/f200-private.o obj/image.o obj/log.o obj/r200.o obj/r200-private.o obj/rs.o obj/stream.o obj/sync.o obj/types.o obj/uvc.o obj/uvc-libuvc.o obj/uvc-v4l2.o obj/uvc-wmf.o obj/libuvc/ctrl.o obj/libuvc/dev.o obj/libuvc/diag.o obj/libuvc/frame.o obj/libuvc/init.o obj/libuvc/stream.o `pkg-config --cflags --libs libusb-1.0` -o lib/librealsense.so
cc examples/c-tutorial-1-depth.c -Iinclude -Llib -lrealsense -lm `pkg-config --cflags --libs glfw3 glu gl` -o bin/c-tutorial-1-depth
lib/librealsense.so: undefined reference to `std::ostream& std::ostream::_M_insert<unsigned long>(unsigned long)'
lib/librealsense.so: undefined reference to `std::basic_ios<char, std::char_traits<char> >::init(std::basic_streambuf<char, std::char_traits<char> >*)'
lib/librealsense.so: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, unsigned long, std::allocator<char> const&)'
lib/librealsense.so: undefined reference to `std::ctype<char>::_M_widen_init() const'
lib/librealsense.so: undefined reference to `std::thread::join()'
lib/librealsense.so: undefined reference to `operator delete(void*)'
lib/librealsense.so: undefined reference to `VTT for std::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >'
lib/librealsense.so: undefined reference to `std::ios_base::~ios_base()'
lib/librealsense.so: undefined reference to `std::basic_istringstream<char, std::char_traits<char>, std::allocator<char> >::~basic_istringstream()'
lib/librealsense.so: undefined reference to `VTT for std::basic_istringstream<char, std::char_traits<char>, std::allocator<char> >'
lib/librealsense.so: undefined reference to `__cxa_end_catch'
lib/librealsense.so: undefined reference to `std::basic_ofstream<char, std::char_traits<char> >::~basic_ofstream()'
lib/librealsense.so: undefined reference to `__cxa_allocate_exception'
lib/librealsense.so: undefined reference to `std::string::reserve(unsigned long)'
lib/librealsense.so: undefined reference to `std::basic_istream<char, std::char_traits<char> >& std::operator>><char, std::char_traits<char>, std::allocator<char> >(std::basic_istream<char, std::char_traits<char> >&, std::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
lib/librealsense.so: undefined reference to `std::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::~basic_ostringstream()'
lib/librealsense.so: undefined reference to `std::runtime_error::runtime_error(std::string const&)'
lib/librealsense.so: undefined reference to `std::string::compare(char const*) const'
lib/librealsense.so: undefined reference to `std::ios_base::ios_base()'
lib/librealsense.so: undefined reference to `std::cout'
lib/librealsense.so: undefined reference to `vtable for std::basic_ios<char, std::char_traits<char> >'
lib/librealsense.so: undefined reference to `std::locale::locale()'
lib/librealsense.so: undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*)'
lib/librealsense.so: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::string const&)'
lib/librealsense.so: undefined reference to `std::basic_filebuf<char, std::char_traits<char> >::open(char const*, std::_Ios_Openmode)'
lib/librealsense.so: undefined reference to `std::ostream& std::ostream::_M_insert<double>(double)'
lib/librealsense.so: undefined reference to `std::basic_ifstream<char, std::char_traits<char> >::~basic_ifstream()'
lib/librealsense.so: undefined reference to `std::string::insert(unsigned long, char const*, unsigned long)'
lib/librealsense.so: undefined reference to `std::string::append(char const*, unsigned long)'
lib/librealsense.so: undefined reference to `std::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream(std::_Ios_Openmode)'
lib/librealsense.so: undefined reference to `std::string::_M_leak_hard()'
lib/librealsense.so: undefined reference to `std::ostream::flush()'
lib/librealsense.so: undefined reference to `typeinfo for std::exception'
lib/librealsense.so: undefined reference to `std::__throw_out_of_range(char const*)'
lib/librealsense.so: undefined reference to `std::ios_base::Init::~Init()'
lib/librealsense.so: undefined reference to `std::ostream::put(char)'
lib/librealsense.so: undefined reference to `std::ostream& std::ostream::_M_insert<void const*>(void const*)'
lib/librealsense.so: undefined reference to `std::string::append(std::string const&)'
lib/librealsense.so: undefined reference to `std::basic_filebuf<char, std::char_traits<char> >::close()'
lib/librealsense.so: undefined reference to `std::condition_variable::~condition_variable()'
lib/librealsense.so: undefined reference to `typeinfo for std::logic_error'
lib/librealsense.so: undefined reference to `vtable for std::basic_streambuf<char, std::char_traits<char> >'
lib/librealsense.so: undefined reference to `typeinfo for std::runtime_error'
lib/librealsense.so: undefined reference to `__cxa_pure_virtual'
lib/librealsense.so: undefined reference to `std::basic_filebuf<char, std::char_traits<char> >::~basic_filebuf()'
lib/librealsense.so: undefined reference to `std::__throw_bad_function_call()'
lib/librealsense.so: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::string const&, unsigned long, unsigned long)'
lib/librealsense.so: undefined reference to `std::__throw_bad_cast()'
lib/librealsense.so: undefined reference to `std::string::assign(char const*, unsigned long)'
lib/librealsense.so: undefined reference to `std::logic_error::logic_error(std::string const&)'
lib/librealsense.so: undefined reference to `std::thread::_M_start_thread(std::shared_ptr<std::thread::_Impl_base>)'
lib/librealsense.so: undefined reference to `vtable for std::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >'
lib/librealsense.so: undefined reference to `std::string::_Rep::_M_dispose(std::allocator<char> const&)'
lib/librealsense.so: undefined reference to `std::runtime_error::~runtime_error()'
lib/librealsense.so: undefined reference to `std::condition_variable::notify_one()'
lib/librealsense.so: undefined reference to `vtable for std::basic_ifstream<char, std::char_traits<char> >'
lib/librealsense.so: undefined reference to `vtable for std::basic_istringstream<char, std::char_traits<char>, std::allocator<char> >'
lib/librealsense.so: undefined reference to `__gxx_personality_v0'
lib/librealsense.so: undefined reference to `std::basic_ios<char, std::char_traits<char> >::clear(std::_Ios_Iostate)'
lib/librealsense.so: undefined reference to `std::string::find(char const*, unsigned long, unsigned long) const'
lib/librealsense.so: undefined reference to `__cxa_begin_catch'
lib/librealsense.so: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&)'
lib/librealsense.so: undefined reference to `__cxa_rethrow'
[... 11 lines cut ...]
lib/librealsense.so: undefined reference to `std::string::swap(std::string&)'
lib/librealsense.so: undefined reference to `std::__basic_file<char>::~__basic_file()'
lib/librealsense.so: undefined reference to `std::istream::operator>>(int&)'
lib/librealsense.so: undefined reference to `VTT for std::basic_ifstream<char, std::char_traits<char> >'
lib/librealsense.so: undefined reference to `std::chrono::_V2::system_clock::now()'
lib/librealsense.so: undefined reference to `std::string::_Rep::_S_empty_rep_storage'
lib/librealsense.so: undefined reference to `std::chrono::_V2::steady_clock::now()'
lib/librealsense.so: undefined reference to `std::basic_ofstream<char, std::char_traits<char> >::basic_ofstream()'
lib/librealsense.so: undefined reference to `std::string::assign(std::string const&)'
lib/librealsense.so: undefined reference to `std::logic_error::~logic_error()'
lib/librealsense.so: undefined reference to `vtable for std::basic_filebuf<char, std::char_traits<char> >'
[... cut ...]
lib/librealsense.so: undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::__ostream_insert<char, std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*, long)'
lib/librealsense.so: undefined reference to `__cxa_free_exception'
lib/librealsense.so: undefined reference to `std::__throw_logic_error(char const*)'
collect2: error: ld returned 1 exit status
make: *** [bin/c-tutorial-1-depth] Error 1

Do anybody knows what's the matter and how to solve this issue? Thanks!

ddiakopoulos, 2016-05-30:
Hi @wllw210 this doesn't seem to be related to librealsense. It looks like your compiler might be borked? Maybe g++ is symlinked to gcc which which doesn't invoke `-lstdc++`

wllw210, 2016-05-30:
Hi ddiakopoulos,
Great thanks for your reply. I thought it was a error related librealsense(some symbol cann't find).
I solved the error based on your reply about g++. Thanks!

closed 2016-05-30 by ddiakopoulos

iindex, 2017-10-23:
How did you resolve the issue with such compilation error? I have similar, but compiling another sources, which I have not developed.