realsenseai/librealsense #158: No device detected.

wllw210, 2016-05-30:
Hi,
when I execute a file in bin directory such as cpp-multicam, I got the error:
No device detected. Is it plugged in?
It mean my device SR300 (I have plugged SR300 into a RIO USB 3.0 port) didn't be recognized by the system. But I don't know why. My dmesg about uvcvedio.ko seems OK:
[  146.431043] media: Linux media interface: v0.10
[  146.458557] Linux video capture interface: v2.00
[  146.493974] uvcvideo: module verification failed: signature and/or required key missing - tainting kernel
[  146.494736] usbcore: registered new interface driver uvcvideo
[  146.494739] USB Video Class driver (1.1.1) 

Does anybody know the reason why my SR300 cann't be recognized? Thanks!

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  ? | 
| Operating System & Version | Ubuntu 14.04 LTS  |
| Kernel Version (Linux Only)  |  4.4.0 |
| Build System |  Makefile |

ddiakopoulos, 2016-05-31:
What is the dmesg output when you plug the device in?

ddiakopoulos, 2016-06-03:
Closed for lack of new info.

closed 2016-06-03 by ddiakopoulos