realsenseai/librealsense #159: Add API to track R200 camera movement in 3D space with 6 DOF

ds-hwang, 2016-05-31:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | ToT  | 
| Operating System & Version | Ubuntu 16.04   |
| Kernel Version (Linux Only)  | v4.4  |
| Build System | Makefile | 

[Intel R200 introduction](https://software.intel.com/en-us/articles/realsense-r200-camera) says
> The R200 tracks the camera movement in 3D space with 6 DOF (degrees of freedom). 3 Degrees from the Front/Back/Up/Down/Left/Right and 3 degrees from Yaw/Pitch/Roll movements.

However, `librealsense` doesn't provide the API.

ddiakopoulos, 2016-05-31:
Hi @ds-hwang as noted in the Readme of librealsense, this project is _not_ the RealSense SDK and doesn't contain the middleware stack (scene reconstruction, 6dof, background segmentation, etc). It's lower level and meant to be used directly be algorithms developers. Hope that helps!

closed 2016-05-31 by ddiakopoulos

ds-hwang, 2016-05-31:
Ah, thx for explaining. I thought the camera has IMU sensor, rather than guessing 6dof from depth data. I agree.

teknotus, 2016-06-01:
I think all of the devices with integrated R200 also have a built in IMU so
it would be redundant to put one in the camera module.

ds-hwang, 2016-06-02:
@teknotus indeed, at least, R200 has IMU sensor, according to [intel article](https://software.intel.com/en-us/blogs/2016/01/28/new-realsense-v8)
As the 6dof data comes from IMU sensor, not estimation, it makes sense to expose the data via API. I guess ROS community really wants the IMU data.

ddiakopoulos, 2016-06-02:
@teknotus @ds-hwang Bad documentation.

The R200 does **not** have any IMU, nor the F200, nor the SR300.

ds-hwang, 2016-06-02:
@ddiakopoulos ok, thx for verifying. I saw several intel realsense docs mentioning IMU sensor. Someone continuously says not true..

zubair123khan, 2017-03-07:
hi all, can any one suggest me any person tracking algorithm for ros using R200 camera
thanks