realsenseai/librealsense #161: Examples not working on Ubuntu 16.04?

mgcrea, 2016-05-31:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0 | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.4.0-22-generic  |
| Build System | Makefile | 

Just followed the starting guide today to try the examples. I'm encountering very weird output (but I'm not sure this is normal).

Basically when running `bin/cpp-tutorial-2-streams`, I get a red blinking square on top left, camera video on top right and all black at the bottom. If I move my hand in front of the camera I can get one pixel switching white on the lower part.

```
# bin/cpp-tutorial-2-streams 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense SR300
    Serial number: 610201001778
    Firmware version: 3.15.0.0
Device does not provide infrared2 stream.
```

There is nothing that remotely looks usable. Thought a firmware update would fix it but no, same behavior.

Is my camera broken?

ddiakopoulos, 2016-05-31:
16.04 is a known working configuration with librealsense +SR300. Could you downgrade the firmware as shown in https://github.com/IntelRealSense/librealsense/issues/151? 

Our testing app usually is `cpp-capture` -- I'm not surprised cpp-restest is kinda broken or that `cpp-config-ui` has a few warnings.

mgcrea, 2016-05-31:
Looks like I got back some more noise on the black part of `bin/cpp-tutorial-2-streams`, still get the blinking red box and now if I put my hand in front of the camera I get some curved lines on the bottom of the screen.

```
bin/cpp-tutorial-2-streams 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense SR300
    Serial number: 610201001778
    Firmware version: 3.10.10.0
Device does not provide infrared2 stream.
```

Not sure what is expected for `cpp-capture` but I get the normal video feed on top-left, a blinking red/blue image on the top right, and a black box on the bottom (with some curved line if I put my hand right in front).

Would a picture help?

mgcrea, 2016-05-31:
Got you a video: https://www.youtube.com/watch?v=HdckzdrRJh4

ddiakopoulos, 2016-05-31:
Could you try it on a Win 8 or Win 10 machine with librealsense?

mgcrea, 2016-06-01:
Does look better: https://www.youtube.com/watch?v=5sgtMElUHbo

Is it the correct output? Is Ubuntu 16.04 supported?

ddiakopoulos, 2016-06-02:
@mgcrea looks great to me. Yeah, 16.04 is running on quite a few of our machines here (and many reports in the wild), so seems like there was an installation issue. That or your Win10 machine is a different hardware issue? Normally for installation issues I recommend re-running the patch script (noting if anything fails, and remembering to `sudo modprobe uvcvideo` after). Also some `demsg` output before/after you plug in a camera would also be helpful.

ddiakopoulos, 2016-06-16:
Closing for lack of new info (suspect this is a common patching issue we see on Ubuntu). Feel free to reopen if you require further support.

closed 2016-06-16 by ddiakopoulos

gsk2286, 2016-10-19:
We recently got the Intel Realsense SR300.We too encountered the same problem that not able to capture the depth data ,but were able to capture RGB frames.But with camera F200 we were able to capture both depth and RGB frames.

We even updated the kernel to Updated 4.4 Stable Kernel.
Camera Model	 SR300
Firmware Version	3.15.0.0
Operating System & Version	Ubuntu 14.04 
Kernel: 4.4.0-38-generic

We tried the bin/cpp-pointcloud but getting blank image but no error displayed,but same works well with F200 camera.Any solution for this?

I guess the problem could be patching the uvc module (uvcvideo.ko).

gsk2286, 2016-10-20:
@mgcrea .I tried to patch the uvc module (uvcvideo.ko) using the script in the script folder 
scripts/patch-uvcvideo-ubuntu-mainline.sh. This worked for me and was able to capture the depth stream from SR300 camera.

sbarkby, 2016-11-18:
Camera Model: SR300
Firmware Version: 3.17.0.0
Operating System & Version: Ubuntu 16.04
Kernel: 4.4.0-47-generic

I also encountered this issue a few weeks ago. After a couple of repeated attempts I managed to resolve the red flashing depth stream issue with the Ubuntu 16.04.xx (Kernel 4.4) Video4Linux backend patch.

It's worth noting though that this issue can resurface. When I left the office yesterday the SR300 was working but when I booted it up in the morning the red flashes were back (I could find no evidence of any automatic updates to the system that could have caused the driver to fail). Fortunately running the Video4Linux backend script again has the SR300 back up and streaming fine.