realsenseai/librealsense #163: R200 fails with librealsense after firmware upgrade

mattswoboda, 2016-06-01:
RS200 / Windows 10 / VS2013 

I've been having a few problems getting the R200 to work properly with LibRealsense. Even in the examples the camera would sometimes fail, throwing exceptions in the lib and requiring to be replugged to make it work again. It generally works, the CPPCapture sample shows the colour, depth and infrared images, but it doesn't feel stable at all. The same problems occur in the integration I've done with our software.

So I installed the latest DCM - Build Number: 2.1.24.9476 - and let it upgrade the firmware. And then the problems really started.. in LibRealsense it either doesn't recognise the camera at all, recognises the streams but hangs on the first attempt to get a buffer, or (best of all) bluescreens the PC when it tries to get a buffer.
Worse, it seems like that upgrade is a one way street. I can't find any way to uninstall the DCM and roll back to an older one, or any way to get back to the state I had before when the camera at least did something. 

Any advice much appreciated. We're integrating the cameras with our software (to go alongside Kinect1 and Kinect2) - but this is currently looking like a dead end.

fabianbl, 2016-06-01:
I don't think it's a problem with the firmware. I also updated it yesterday with DCM (build 2.1.24.9476). I tested some of the example apps (e.g. the cpp-capture app) on both Windows 8.1 with VS15 (Community) and on Ubuntu 14.04 and had very few crashes, I don't remember actually if there were any crashes at all. And the camera is always properly recognized.

The only example that I found not to be too stable yet is the cpp-config-ui program which always crashes on both operating systems when I try to change the R200_AUTO_EXPOSURE_MEAN_INTENSITY_SET_POINT, I will try to have a closer look at what's actually the problem there. The RealSense SDK also works fine most of the time.

Could you already test your camera on another computer to figure out if it's rather a problem with the computer or with the camera?

mattswoboda, 2016-06-01:
If it was a problem with either the computer or the camera, why did it only break after I installed the new DCM?

ddiakopoulos, 2016-06-01:
Hi @mattswoboda -- sorry for all your troubles! In general, debugging our cameras comes down to 2 or 3 things. Very rarely it's a problem with the software in this repo, and even more rarely is it a H/W issue. 

  1. Make sure it's on a compatible system (where compatibility is a function of supported USB host controller chipset that provides enough power)
  2. Driver installation issues (In 99% of cases this means Ubuntu/Linux)
  3. Firmware. We don't track DCM version number (... and that ecosystem is largely a disaster because of all the reasons you mention related to the installation/upgrade process).

Please shoot me an email @ dimitri.diakopoulos@intel.com and I'll send you the direct F/W update binary for Windows. That'll be a good place to start. There definitely should be a mechanism to uninstall the DCM, so definitely do that as well!

mattswoboda, 2016-06-02:
@ddiakopoulos : Mail sent - thanks!

ddiakopoulos, 2016-06-02:
Got it. Will follow up soon... need to find the installer binary.

RickBlacker, 2016-06-16:
What became of this issue?  Was the problem resolved?

mattswoboda, 2016-06-17:
Not yet - I still need the firmware file to flash the device. Any help gratefully received.

jlgarcia75, 2017-10-03:
Hello,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75