realsenseai/librealsense #166: Sync, resolution, misc.

jmaye, 2016-06-03:
Hi,

I remember from one of my issues that it was possible to do hardware clock synchronization on the Windows SDK with the SR300. I've just tried it and it does not seem to work properly though. Any hints with that? Is this feature not available yet?

Let's say I don't have clock sync., then I have to do it myself in software I guess. I believe this is what you do in librealsense in the sync.cpp file? Just to be sure, the timestamp outputted from the get_frame_timestamp comes from the sensor itself or is it just an OS timestamp when the frame arrives in the USB layer?

About resolution, what is the best strategy here? I've seen that in all your examples you use 640x480 for color stream, is this just random or is it recommended? I tried with the F200/SR300 and it seems to go to 1080p. I read somewhere that the intrinsics might not be good, is it still true? I just outputted them for these resolutions and they look good to me. How was this calibrated in the end? At all resolutions or at just one and then scaled?

Why I'm asking those questions, my goal is to make the best out of these cameras (either F200 or SR300) in terms of resolution, framerate, accuracy, and critically synchronization (in order to texture a mesh for instance).

Lastly, I've seen a bunch of options/configurations in the cpp-config-ui, all the part concerning color stream is standard, but the rest is not that well documented. Laser power is easy to understand, however whether I have to tune it, no idea. For the "motion vs range", increasing it seems to reduce the framerate and augment the range. "Filter options" looks like a smoothing parameter, at "'0" we see the stripes for instance. By default, it is at "5". What is done here, a median filter? "Confidence threshold" seems to filter out points based on confidence or is it also affecting smoothing? "Depth control presets" do not work and "IVCAM preset" adjusts the above parameters according to some profile. "Accuracy" I don't see quite a difference. I'm a bit hesitant about all these settings, should I leave it to default or adjust them. Can you comment on that?

Thanks for all your efforts and keep on the good work. I'd be glad to help if I can.

ddiakopoulos, 2016-06-03:
All the RealSense sensors are a bit different so I'll only talk about the SR300 here.

- Frame timestamps are indeed generated on the camera using an on board hardware counter (not a real time clock or anything fancy; but we definitely try to avoid OS timestamps as much as possible)
- The first few pixels of each image are actually data with this counter value (and they periodically overflow, so the software needs to track them as well)
- The RGB and IR imager are not exposed together and have highly independent framerates (depending on filter settings, exposure, etc)
- The purpose of the librealsense sync.cpp is to provide to provide sets of depth/ir/rgb frames with similar timestamps
- The SDK (and future versions of librealsense) will provide a working version of HW sync, which SR300 supports
- HW sync depends on some per-frame metadata structs as part of the UVC protocol. These are easy to access on Windows, but require yet more kernel patches on Linux. This is presently not implemented in our library.
- HW sync actually does attempt to trigger the imagers at the same time, and provides per-frame (via the struct) into about timing and exposure.
- Depth Control Presets only affect the R200 which you've already noted
- The RealSense SDK does provide a good amount of documentation on the available camera controls (https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/property_device_pxccapture.html) but librealsense sometimes has different identifiers for them.
- The SR300 is not per-unit calibrated, and so all calibration info (for all resolutions, I think) is based on a model. It _might_ be possible to get a better color camera calibration manually, but I don't recommend it.
- The nominal/best resolution for SR300 is 640x480 @ 60fps. The color imager is capable of 1080p @ 30fps. 
- Any data related to accuracy is highly dependent on both camera settings and environment/object

Pinging @dorodnic for more info I didn't address here. He's recently taken over lead development and also comes from the SR300 team, so his knowledge of it greatly exceeds mine.

ghindman, 2016-06-03:
Do all of the cameras have an onboard hardware counter, or just SR300, and is that counter shared across all the sensors - ie., an IR frame and a color frame with the same timestamp were stamped within the same time-domain and were taken at the same time?

ddiakopoulos, 2016-06-03:
R200 and F200 also have hardware counters. Yes, all the sensors use the same time domain, but "taken at the same time" is a question of synchronization. SR300 is the only camera capable of hardware synchronizing exposure times of the IR/RGB camera. Everything else requires a fuzzy-matching timestamp-based model.

ghindman, 2016-06-03:
Understood - by "same time" i didn't mean commanding the sensors to capture simultaneously, but rather two independent frames with the same timestamp were both captured withing the same increment of the onboard counter and therefore relatively time-aligned.

teknotus, 2016-06-03:
> HW sync depends on some per-frame metadata structs as part of the UVC
protocol. These are easy to access on Windows, but require yet more kernel
patches on Linux. This is presently not implemented in our library.

I've been focused on Linux kernel driver development for a few months. If
you can tell me which bytes you need I can probably make a patch to export
them through the new per frame metadata api.

jmaye, 2016-06-06:
Thanks all for the comments! I will look deeper into it. For sure, if we can get the HW sync feature from SR300 on the Linux port that would be great. As I mentioned, I think HW sync does not work either on the Win SDK, a colleague of mine told me that no frame arrives in the end with this mode. What I also found strange is that the timestamps reported by my colleague on the Win SDK look totally different than on a Linux machine, they look more like ticks from the epoch. While I had stuff close to zero, he had much higher values.

ddiakopoulos, 2016-06-06:
Yeah, I should probably mention that the synchronization logic for the RealSense Windows SDK and librealsense are *very* different. If it's ok with you, I'll flag this issue as a feature request for HW Sync support for SR300.

jmaye, 2016-06-06:
Great, thanks! One last remark, let's say this HW sync is done, the Linux port is fully functional in terms of low-level driver? There is no particular reason of using the Windows SDK except for higher-level functionalities?

jmaye, 2016-06-08:
In the Windows SDK, is this the way to go to enabling strong sync?:

// Select the color and depth streams
sm->EnableStream(PXCCapture::STREAM_TYPE_COLOR,0,0,0,PXCCapture::Device::STREAM_OPTION_STRONG_STREAM_SYNC);
sm->EnableStream(PXCCapture::STREAM_TYPE_DEPTH,0,0,0,PXCCapture::Device::STREAM_OPTION_STRONG_STREAM_SYNC);

When we do that on the SR300, the callbacks are not triggered, hence no data arrives.

It's a bit fuzzy right now in these threads. is this HW sync supposed to work on Win SDK with SR300 or is it work in progress? I understand that this is not yet working on Linux and @teknotus has proposed to work on it.

jmaye, 2016-06-08:
Ah, and the Win SDK timestamps are related Windows epoch? I get stuff like 131098555995823238, which would correspond to current date according to http://www.epochconverter.com/ldap. I guess that it just takes the Windows timestamp at the start of the recording and then increment with HW timestamps.

mkhansen-intel, 2017-01-07:
@ddiakopoulos and @dorodnic  - So I see timestamp offsets of ~5 up to ~15 ms when running the SR300, but not with the R200. Is this to be expected?

Also in your comments above you mentioned adding HW Sync support for SR300 in Linux as a feature request, has that been done yet or if not, when would we expect it?

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

ddiakopoulos, 2017-10-07:
Hi @jlgarcia75 -- i think for best support, issues should be responded directly in the issue queue for this repository.

closed 2017-10-16 by jlgarcia75