realsenseai/librealsense #167: SR300 support

nlyubova, 2016-06-06:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.10.10.0  | 
| Operating System & Version |  Ubuntu 14.04 |
| Kernel Version (Linux Only)  |  4.4.0-040400-generic |
| Build System |   Makefile | 

Hello, I'm using your driver with a SR300 model and I wonder if you could help me to fix the color image (the depth is fine). I'm attaching a color image when capturing a cola and smarties. All colors are saturated, what can be a problem? what is the difference in color format/conversion for R200 and SR300?

[image]

Thank you,

linknum23, 2016-06-06:
I had to set the color_saturation to 43. RGB image looks fine after that. My guess is that the default of 0 works fine for the R200 but not for the SR300.

nlyubova, 2016-06-06:
hmm.. it does not work for me. which mode do you use? manual?

linknum23, 2016-06-06:
Yeah the mode is set to manual by the ros driver I'm using. Here is my camera configuration: [image]

nlyubova, 2016-06-06:
oh, sorry, now it works ! Thank you!

closed 2016-06-07 by ddiakopoulos