realsenseai/librealsense #169: R220 Stream Documentation

mdhorn, 2016-06-07:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version | ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.4.0-22-generic |
| Build System |  Makefile | 

The only published documentation I can find on valid stream resolutions and viable combinations is located on this web page:
 https://software.intel.com/en-us/articles/realsense-r200-camera
Available Resolutions
@60FPS, depth at 320x240,, color can be 640x480
@60FPS, depth at 480x360, color can be 320x240 or 640x480
@30FPS, depth at 320x240,, color can be 640x480, 1280x720 or 1920x1080
@30FPS, depth at 480x360, color can be 320x240, 640x480, 1280x720, or 1920x1080

However, the 1280x720 stream does not exists on the R200 with FW 1.0.72.06.

Is there better, or more current documentation available for the R200? Also for the other RealSense cameras?

teknotus, 2016-06-07:
For each of the cameras in /dev/video you can get a complete list of
resolutions using a Linux video tool.

v4l2-ctl --list-formats-ext -d /dev/video

digitologist, 2016-06-07:
The DCM release notes include this chart, is it what you're looking for?
http://registrationcenter-download.intel.com/akdlm/irc_nas/9078/release_notes_RealSense_SDK_2016_R2.pdf

[image]

mdhorn, 2016-06-07:
Yes, exactly what I wanted for reference.
How current is this document?
I ask because the R200 with the FW I listed above doesn't support any 1280x720 color streams.

ddiakopoulos, 2016-06-13:
@mdhorn all documentation is old. The firmware and spec documents are never in sync. The cross-section of validated modes is indeed smaller than what the firmware provides as well. Because of this, librealsense hardcodes streaming modes most commonly used (and many uncommon ones) because we were routinely finding modes that didn't work as published by the UVC descriptors on the camera firmware (as noted by @teknotus above).

teknotus, 2016-06-13:
So the descriptors don't always match what formats the camera will actually
stream? Does the camera report back unsupported mode, or is it a worse
error?

mdhorn, 2016-06-13:
If you ask the camera FW what it is supports via librealsense (get_stream_mode), it only reports valid modes.

But, if you use 'v4l2-ctl --list-formats-ext -d /dev/video2', unsupported modes are reported.

ddiakopoulos, 2016-06-13:
Correct. v4l2-ctl actually reaches out to the camera and parses the UVC desciptors. librealsense consciously does something a little more 'dangerous' by hardcoding the subset of modes that are almost guaranteed to work. We've discussed it internally a few times and librealsense _may_ eventually support a get_stream_mode that actually asks the camera first.

ddiakopoulos, 2016-06-13:
@mdhorn does v4l2-ctl report a 1280x720 mode still? Just checking to confirm it wasn't removed in the firmware side. We should probably support it in librealsense, but we will have to interpolate intrinsics.

mdhorn, 2016-06-13:
1280x720, as well as several other "interesting" modes are reported:

```
$ v4l2-ctl --list-formats-ext -d /dev/video2
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'pRAA'
	Name        : 10-bit Bayer RGRG/GBGB Packed
		Size: Discrete 2400x1081
			Interval: Discrete 0.033s (30.000 fps)
```

ddiakopoulos, 2016-06-16:
@mdhorn okay, so then we need to add the 1280x720 color stream to the internal R200 mode list. It will complicated the generation of stream intrinsics. There's an existing issue (#135) to implement this functionality. Closing this out and will track #135.

closed 2016-06-16 by ddiakopoulos