realsenseai/librealsense #170: R200 Depth Options

fabianbl, 2016-06-08:
I would like to test different settings for the RS_OPTION_R200_DEPTH_CONTROL_... parameters, but I could not find any explanation of what they do. Is there some documentation about which algorithm is used for disparity map generation / stereo matching and how I can affect it with these parameters?

ddiakopoulos, 2016-06-13:
Hi @fabianbl - Pinging @leonidk who can offer some more insight.

leonidk, 2016-06-14:
We're not planning to disclose any additional details at the moment but I'll reopen this issue if we make more information available. At the moment the user should just select the best preset for their use from rs_utils.h

closed 2016-06-14 by leonidk

fabianbl, 2016-06-14:
Ok, no problems. Thanks for your reply.

l3s777, 2016-08-10:
hi @fabianbl  @ddiakopoulos  can I have some more info about that as well?
I am trying to improve the image I am receiving from R200 in "depth"

thanks

l3s777, 2016-08-11:
I used it like this:
dev->set_option(rs::option::r200_emitter_enabled, 1);

it seems to work.

fabianbl, 2016-08-11:
Yes, this is how you can change the parameters. If you want to do it dynamically in a GUI, you can use the [cpp-config-ui](https://github.com/IntelRealSense/librealsense/blob/master/examples/cpp-config-ui.cpp) example which does this for all available parameters.

leonidk, 2017-05-17:
@l3s777 @fabianbl We've published some additional information here
https://arxiv.org/abs/1705.05548

fabianbl, 2017-05-17:
@leonidk Nice, thanks for the notice.