realsenseai/librealsense #171: Cleaning disconnecting camera

realsensegithub, 2016-06-09:
| Required Info  |   |
|---|---|
| Camera Model | R200| 
| Firmware Version |  1.0.72.06 | 
| Operating System & Version | Windows 10  |
| Build System |  VS2013 | 

I often run into problems where my program crashes and the cameras become unresponsive. It can take several attempts at unplugging and replugging them in as well as closing down and opening my application to finally get them to capture again. This happens quite sporadically although they still appear in device manager.

Is there a good way to cleanly shutdown the cameras so that the next time I run my application I can make sure the cameras will start? Presently I have this issue where I can grab an image from the camera but I cannot do anything with it, I can reassign it to an OpenCV mat and the image data is there but calling .clone or any other operation to it causes OpenCV to throw an exception. I believe this might be to do with the previous run of the program that crashed.

ddiakopoulos, 2016-06-13:
Lots of little issues here. If you are using the C++ API (rs.hpp) then the context should shut down the cameras cleanly on exit.

Not sure about your OpenCV issue. Pointers to librealsense frames are only valid until the next call to wait_for_frames(). If you want to keep the mat around, I recommend a memcpy.

On particular host controller chipsets / CPUs, we've noticed that some RealSense cameras are quite unstable (hardware issues out of our control, unfortunately). I always recommend trying different physical ports on the system, or with a powered USB 3 hub.

ddiakopoulos, 2016-06-25:
Closed for lack of new info.

closed 2016-06-25 by ddiakopoulos