realsenseai/librealsense #172: SR300: cpp-config-ui crash when setting MOTION_VERSUS_RANGE

wangji1, 2016-06-13:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.10.10.0  | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System | Makefile | 

In cpp-config-ui, there are 4 options for MOTION_VERSUS_RANGE: 
SR300_AUTO_RANGE_ENABLE_MOTION_VERSUS_RANGE;
SR300_AUTO_RANGE_MIN_MOTION_VERSUS_RANGE;
SR300_AUTO_RANGE_MAX_MOTION_VERSUS_RANGE;
SR300_AUTO_RANGE_START_MOTION_VERSUS_RANGE;
When adjust the options, the tool will be crashed.
| RealSense error calling rs_set_device_option(device:0x298ec40, option:SR300_AUTO_RANGE_MIN_MOTION_VERSUS_RANGE, value:595):
    opcodes do not match |

dorodnic, 2016-06-14:
Options starting with SR300_AUTO_RANGE are meant to be configured to correct values through rs_apply_ivcam_preset in rsutils.h. 
Trying to set those manually is likely fail, resulting in config.ui crash.

closed 2016-06-14 by dorodnic

wangji1, 2016-06-15:
In rs_apply_ivcam_preset , it lists the default values for arr_options, right? 
If I change the arr_values, should it be correctly showed on config.ui?