realsenseai/librealsense #173: SR300: cpp-config-ui crash when setting LASER

wangji1, 2016-06-13:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.10.10.0  | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System |  Makefile | 

Playing the cpp-config-ui, when adjust SR300_AUTO_RANGE_MIN_LASER, or SR300_AUTO_RANGE_MAX_LASER, or SR300_AUTO_RANGE_START_LASER, the tool will crash.
RealSense error calling rs_set_device_option(device:0x163cc40, option:SR300_AUTO_RANGE_MIN_LASER, value:-3277):
    opcodes do not match

m0ose, 2016-06-21:
I have the same problem on ubuntu 16.04 LTS

halton, 2016-06-28:
Reproducible for me on Ubuntu 16.04 as well. Besides the above three options, the `SR300_AUTO_RANGE_MIN_MOTION_VERSUS_RANGE` and `SR300_AUTO_RANGE_MAX_MOTION_VERSUS_RANGE` cause crash as well.

Wollimayer, 2016-07-04:
I think , this is coused by the #191 issue.

dorodnic, 2016-09-06:
set_option should not crash demos now. Please review the comment in #191 on SR300 AUTO_RANGE options.

closed 2016-09-06 by dorodnic