realsenseai/librealsense #175: Ubuntu 16.04: Invalid depth stream

alperaydemir, 2016-06-14:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  3.10.10.0 | 
| Operating System & Version | Ubuntu 16.04 |
| Kernel Version (Linux Only)  | 4.4.0-24-generic #43-Ubuntu  |
| Build System | Makefile | 

Hi,

On a fresh Ubuntu 16.04 installation, running the example programs I get the below screenshot. Depth and Infrared streams seem to be invalid. The depth window keeps flashing. I have a pretty good USB3 card, one of the industrial ones that has a separate controller behind each port. Any idea how to fix it?

[image]

ddiakopoulos, 2016-06-14:
Hi @alperaydemir -- these issues seem consistent with an issue in the patched uvcvideo driver. Can you re-run the patching script for 16.04 again (and make sure to `sudo modprobe uvcvideo` at the end).

bloyl, 2016-06-15:
@ddiakopoulos - I am seeing a similar issue using an F200 with the stock 16.04 kernel (4.4.0-24-generic #43-Ubuntu SMP). That camera seems to work fine with librealsense in windows 8.1.

Perhaps I'm misunderstanding the installation guide but the guide says:

> For Ubuntu 16.04 -- No kernel patch is needed for the R200 camera
is this true for the other cameras as well?

> Kernel in 16.04.xx
> No action required. Ubuntu 16.04 beta2 and newer already contains the patch

What patch should be used?

FYI my camera is running Firmware version: 2.60.0.0
and my kernel is 4.4.0-24-generic #43-Ubuntu SMP

here is some dmesg output

[  604.620170] usb 6-2.1: Product: Intel(R) RealSense(TM) 3D Camera (Front F200)
[  604.620172] usb 6-2.1: Manufacturer: Intel(R) RealSense(TM) 3D Camera (Front F200)
[  604.620175] usb 6-2.1: SerialNumber: 014150018607
[  604.623465] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) 3D Camera (Front F200) (8086:0a66)
[  604.626407] input: Intel(R) RealSense(TM) 3D Camer as /devices/pci0000:00/0000:00:03.0/0000:04:00.0/usb6/6-2/6-2.1/6-2.1:1.0/input/input18
[  604.627444] uvcvideo: Unknown video format 5a564e49-2d90-4a58-920b-773f1f2c556b
[  604.627450] uvcvideo: Unknown video format 495a4e49-1a66-a242-9065-d01814a8ef8a
[  604.627455] uvcvideo: Unknown video format 52564e49-2d90-4a58-920b-773f1f2c556b
[  604.627460] uvcvideo: Unknown video format 49524e49-2d90-4a58-920b-773f1f2c556b
[  604.627464] uvcvideo: Unknown video format 49564e49-57db-5e49-8e3f-f479532b946f
[  604.627469] uvcvideo: Unknown video format 494c4552-1314-f943-a75a-ee6bbf012e23
[  604.627777] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) 3D Camera (Front F200) (8086:0a66)
[  604.630329] uvcvideo: Unable to create debugfs 6-5 directory.
[  604.630591] input: Intel(R) RealSense(TM) 3D Camer as /devices/pci0000:00/0000:00:03.0/0000:04:00.0/usb6/6-2/6-2.1/6-2.1:1.2/input/input19

Thanks
Luke

bloyl, 2016-06-15:
so my issue was an unpatched uncvideo module. there is a patch file scripts/patch-uvcvideo-16.04.simple.sh which does that patching. Note that you will need to enable src repos in your /etc/apt/sources.list.

ddiakopoulos, 2016-06-16:
Hi @alperaydemir any update?

alperaydemir, 2016-06-16:
Hi, yes I confirm my issue is the same as @bloyl. The scripts somehow didn't patch the uvcvideo module and I used a pre-compiled version somebody in the office had, which seemed to do the trick.

The installation instructions are somewhat unclear since they say that no kernel update needed for 16.04 but this isn't my experience. I had to get the 4..4.0-040400 kernel.

Therefore I'm closing this issue but somebody should probably go through in the installation steps and verify what's wrong (or right) there.

closed 2016-06-16 by alperaydemir

ddiakopoulos, 2016-06-16:
Actually one step ahead of you @alperaydemir -- fixed in 9a00ba53458fd1a094aba3c600d5dafc71a4d927. The instructions were indeed wrong.

ghindman, 2016-06-16:
Installation.md now directs global installation of the UVC patch for 16.04.  My understanding was that SR300 required the uvc patch, but R200 did not - is this not correct?  Assuming it is correct would it not be better to note the UVC patch as a dependency for SR300 only?

RickBlacker, 2016-06-16:
Hi, what happens if you run the patch, will it have a negative impact on R200?

mattcurfman, 2016-06-16:
Ubuntu 16.04 cherrypicked the publicly available upstream patch for R200 camera, therefore no kernel patching is required for R200 camera.  See https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1557138

The F200/SR300 cameras still require an additional patch regardless of OS. Maybe somebody would upstream those patches, also?

teknotus, 2016-06-17:
> The F200/SR300 cameras still require an additional patch regardless of
OS. Maybe somebody would upstream those patches, also?

I guess I should probably be the one to do this? The patch in the scripts
directory has my comments so I'm actually an upstream author from Intel as
far as the F200 patch.

ddiakopoulos, 2016-06-17:
@mattcurfman yes, from that cherry-pick request, the R200 (ds4) patch will be available in 16.04 but it is _not_ available as part of the current `linux-lts-xenial` package. The cherrypick exists in the Proposed branch (https://wiki.ubuntu.com/Testing/EnableProposed) and will not be available in the Updates branch until 16.04 LTS gets its first point release (16.04.1, on July 21st 2016 according to https://wiki.ubuntu.com/XenialXerus/ReleaseSchedule)

mambawockeez, 2018-08-16:
Hi i have a really simple question, for your ubuntu 16.04 LTS, are you running it on virtual machine? I used the virtual box but ubuntu even didnt detect my sr300.