realsenseai/librealsense #177: R200 Timeout waiting for frames (Ubuntu, Unity3D)

narfanator, 2016-06-15:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.71.06  | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  |  4.5 |
| Build System | Makefile |

I have a Unity3D plugin written in C++ that uses librealsense. The application built using the plugin runs fine on Mac OS, but gets an rs::error "Timeout waiting for frames" when run on Linux. However, the examples included in the RealSense SDK both build and run fine on that same Linux box.

I'm stumped. What can I do to investigate this?
Thanks!

ddiakopoulos, 2016-06-15:
I'm confused. The RealSense SDK doesn't build or run on Linux -- librealsense is the only option.

narfanator, 2016-06-17:
Sorry, I think of librealsense AS an SDK because you build it, and it's got examples - it's not just a library. LibRealSense is what I've got.

narfanator, 2016-06-20:
It appears that this may be related to streaming the infrared-camera when NOT streaming the color camera. When streaming both the infrared and the color, or just the color / just the depth, I do not get the error. Only when streaming the infrared camera without streaming the color camera do I get this error.

narfanator, 2016-06-21:
I also had to enable both IR streams (infrared and infrared2) in order to actually get images out of either.

ddiakopoulos, 2016-06-25:
Can confirm that it's a known issue that you need to pull both IR streams. Possibly related to firmware?

closed 2016-06-25 by ddiakopoulos