realsenseai/librealsense #179: Streaming at high framerate: fps not accurate?

niccarey, 2016-06-18:
Camera model: SR300
OS: Linux (Ubuntu 14.04, Kernel 4.4)

I have a question about frame access at high speeds (possibly silly; I'm not fluent in C++ and may be missing something obvious)

- I'm trying to store high-resolution colour images at 30fps, and depth images at a rather lower framerate
- These do not have to be synchronised.
- I'm not sure if librealsense allows storage of the native RSSDK format available in the Windows SDK, however we require at least an hour's footage at high framerate, and given the large filesizes associated with this format, it's in any case not feasible for us. 
- So I've written a rather naive program which creates and detaches a new thread to store each incoming frame either as an image or a binary file (depending on stream type), and this in principle works quite well.

However, that said, I'm running into issues at high framerates. Specifically, the command
dev->wait_for_frames 
limits the framerate to about 8 or 9fps (verified by pointing the sensor at an external clock). This is true **even when only a colour stream is enabled,** so I do not think it is a synchronisation issue. It is also not a computational speed issue, because the threaded processing reduces the processing loop time to well under the requisite 33ms (and indeed i have confirmed this, by running the loop on a 'frozen' image frame). So it does not seem to be an problem with the image storage - it's the streaming itself that is not running fast enough. Querying the framerate returns 30fps, as requested - but clearly this is not true in fact. 

Without using wait_for_frames, I can store images at an appropriate framerate as measured by clock time, but the stored frames are either incomplete or totally blank, so I don't think this is a valid solution. We were relying on the high framerate offered by the RealSense to gather high-resolution tracking data in RGB-D, and have only a few weeks per year where this is possible (we are now in the field and the clock has started ticking!) So help/suggestions urgently needed and much appreciated!

niccarey, 2016-06-18:
update: probably solved (see top paragraph)

niccarey, 2016-06-19:
update: seems that the resolution is the problem, I can get 30fps by dropping to 640x480. But according to all documentation I can find, streaming at 1920x1080 in colour at 30fps should be fine?

ddiakopoulos, 2016-06-25:
USB bandwidth could be another factor. Although, I think the primary reason is that the SR300 will aggressively grow exposure values in low-light conditions, lowering the the FPS of the color stream.

ddiakopoulos, 2016-07-31:
Closing out issue for lack of new discussion. Feel free to re-open if you have further questions with your configuration.

closed 2016-07-31 by ddiakopoulos

Jowizo, 2017-04-28:
@ddiakopoulos Is there a way to disable the auto exposure values in low light conditions? That way hopefully keep it fixed on 60fps?