realsenseai/librealsense #180: librealsense Build Problem

won13y, 2016-06-20:
Camera Model is intel realsense R200 in Ubuntu 14.04 

I tried to compile by following commands

    make && sudo make install

But the following error occurs:

    In file included from src/uvc-v4l2.cpp:31:0:
    /usr/local/include/libusb-1.0/libusb.h:937:4: warning: ISO C++ forbids zero-size array ‘    iso_packet_desc’ [-Wpedantic] [0] /* non-standard, but usually working code */
    ^
    src/uvc-v4l2.cpp: In function ‘const char* rsimpl::uvc::get_usb_port_id(const rsimpl::uvc::device&)’:
    src/uvc-v4l2.cpp:409:72: error: ‘libusb_get_port_number’ was not declared in this scope
    std::to_string(libusb_get_port_number(device.usb_device));
                                                                        ^
    make: *** [obj/uvc-v4l2.o] Error 1

| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 |  R200
| Firmware Version |   |   
| Operating System & Version |   | Ubuntu 14.04
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile |

dorodnic, 2016-06-20:
Please check your installation of libusb. Assuming you properly install (libusb)[https://github.com/IntelRealSense/librealsense#installation-guide] and there is no other version on your system, you should have (libusb_get_port_number)[http://libusb.sourceforge.net/api-1.0/group__dev.html#ga14879a0ea7daccdcddb68852d86c00c4] defined in libusb.h:1355. 
According to Travis-CI, master is passing compilation on Linux 14.04.

won13y, 2016-06-20:
Thank you for the reply

Actually I did the following commands in following website:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

 2. Install libusb-1.0 via apt-get

    sudo apt-get install libusb-1.0-0-dev

The following is the result of above commands:

    wkyoun@wkyoun-15Z960-GA70K:~/Downloads/RealSense_ROS/realsense_dist/2.3$ sudo apt-get install libusb-1.0-0-dev
    [sudo] password for wkyoun: 
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    libusb-1.0-0-dev is already the newest version.
    The following packages were automatically installed and are no longer required:
      libntdb1 linux-headers-3.19.0-25 linux-headers-3.19.0-25-generic
      linux-image-3.19.0-25-generic linux-image-extra-3.19.0-25-generic
      linux-signed-image-3.19.0-25-generic python-ntdb
    Use 'apt-get autoremove' to remove them.
    0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.

But, It still produce the same error as I stated above

Is installation of libusb complete?

Would you please let me know how to fix it?

won13y, 2016-06-25:
Thank you for the comment,

it was because of libusb library that was installed before.

I erased it, and it works correctly!

closed 2016-06-25 by won13y